rpng/open_vins #341: Build with ROS2 Humble, Ubuntu 22.04, Docker

ArthurLovekinGecko, 2023-06-08:
Hi, I just installed the OpenVINS project using two methods: Locally with ROS2 Humble, Ubuntu 22.04, and using the Dockerfile_ros2_20_04. These installations went relatively smoothly and so far things work, but I had a few challenges especially with Docker. My experience with each is described below for reference.

My questions are: 
Do you plan on updating the documentation and codebase to work for ROS2 Humble? From my perspective it looks like it would not require many changes. Are there any things that might break if I am using ROS Humble? 

1. Local install on my system that is ROS2 Humble, Ubuntu 22.04
- I followed the [Getting Started with ROS](https://docs.openvins.com/gs-installing.html) guide and the [Simple Tutorial ](https://docs.openvins.com/gs-tutorial.html) Guide. 
- Colcon build works and the algorithm seems to run on the Euroc MAV dataset example. The only issue in the build were some deprecated header file names in ov_msckf. For example: /opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]

2. Install using Docker to build the Dockerfile_ros2_20_04.
- I followed the [Docker Installation Guide](https://docs.openvins.com/dev-docker.html) and the [Simple Tutorial ](https://docs.openvins.com/gs-tutorial.html) Guide.
- I had to add the --privileged flag to the docker run command to get gazebo/rviz to run. 
- I also added --name so the container name is always the same, and --rm so the containers don’t build up as you run this command repeatedly. The final run command is:

docker run -it --rm \
    --name openvins_container \
    --net="host" \
    --gpus all \
    --env="NVIDIA_DRIVER_CAPABILITIES=all" \
    --env="DISPLAY" \
    --env="QT_X11_NO_MITSHM=1" \
    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    --mount type=bind,source=$DOCKER_CATKINWS,target=/catkin_ws \
    --mount type=bind,source=$DOCKER_DATASETS,target=/datasets \
    --privileged \
    ov_ros2_20_04 \
    bash

goldbattle, 2023-06-08:
There are some OpenCV issues I want to also sort out https://github.com/rpng/open_vins/issues/316 and https://github.com/rpng/open_vins/issues/304 but it seems that you were able to get things to work relatively easily. I have not upgraded my own OS yet from 20.04 so have not spent time investigating. One thing is to ensure the project maintains some compile backwards compatibility, which is why there are some issues with headers on the newer versions.

Genozen, 2023-09-29:
Hello, I'm interested in getting started with `ROS 2 Humble` for `OpenVins` . I'm debugging some installation issues on some libraries, but I'm not seeing a guide specificlly for `Humble`.  Can I safely assume that the [Getting Started Guide](https://docs.openvins.com/gs-installing.html) (although only specified up to Galactic) should work?

goldbattle, 2023-10-07:
Yes, humble should work fine.

closed 2023-10-28 by goldbattle