rpng/open_vins #345: How to launch and change topics for monocular visual inertial odometry?

nicko369, 2023-06-20:


nicko369, 2023-06-20:
Also since I'm simuating this in gazebo, do I set the camera parameters and IMU parameters to 0?

goldbattle, 2023-06-20:
If you could provide more details and configuration you are using, that would be great. Make sure to go through the tutorials to understand the package first and what is required as input (inspect the example rosbags).
https://docs.openvins.com/gs-tutorial.html

You need some noises, so you could leave those at 1pixel and maybe the default for eurocmav. But you will need the calibration transform between the camera and IMU sensor you subscribe to.

nicko369, 2023-06-20:
I've been going over the documentation page but to be honest it's not very intuitive.
I'm running openvins through dockers.
I've used both ros1 and ros2 dockerfiles you have provided however I can't even get roscore to start. 

I'd like to run this with a turtlebot waffle pi so I'd be using a pi cam.  
I understand I'd need at least the image_raw and IMU data but I've renamed those topics which is why I'd like to remap them in openvins

nicko369, 2023-06-20:
Basically I have created a simulated world in gazebo
I'd like to use a turtlebot 3 waffle pi which has a pi camera
and I'd like to run openvins through dockers.

nicko369, 2023-06-20:
May I give you some thoughts and suggestions about the guide?

closed 2023-06-20 by nicko369