rpng/open_vins #346: Error building colcon workspace in ROS2 20.04 Docker

nicko369, 2023-06-20:
ov_ws/src/open_vins/ov_eval/../ov_core/src/utils/opencv_yaml_parse.h:28:10: fatal error: opencv2/opencv.hpp: No such file or directory
   28 | #include <opencv2/opencv.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ov_eval_lib.dir/build.make:128: CMakeFiles/ov_eval_lib.dir/ov_ws/src/open_vins/ov_core/src/feat/FeatureInitializer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:339: CMakeFiles/ov_eval_lib.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
---
Failed   <<< ov_eval [1min 19s, exited with code 2]
Aborted  <<< ov_core [4min 2s]                        
Aborted  <<< ov_init [6min 31s]                       
Aborted  <<< ov_msckf [8min 11s]          

Summary: 1 package finished [8min 11s]
  1 package failed: ov_eval
  3 packages aborted: ov_core ov_init ov_msckf
  5 packages had stderr output: ov_core ov_data ov_eval ov_init ov_msckf

goldbattle, 2023-06-20:
I am able to build on my machine and the CI passes. Ensure you have a valid
ROS environment and that it has cv_bridge and/or opencv installed.

nicko369, 2023-06-20:
I've built the image from the dockefile provided 
I'm able to find the opencv4/opencv2 file in /usr/include/opencv4 and have tried adding it to the cmake but I still encounter the error

nicko369, 2023-06-20:
I'm just getting errors and unable to build the workspace in both ros1 and ros2 20.04 dockerfiles
I've used dockers to host other algorithms like nav2 and orb slam 3.
Could you guys please build the files from scratch yourselves and show the commands used?

nicko369, 2023-06-20:
The dockerfiles for ros1 and ros2 20.04 are very broken.
They do not work out of the box at all.

I've built the dockerfile 
```
docker run -td --privileged --net=host --ipc=host \
    --name="openvins" \
    --gpus=all \
    -e "DISPLAY=$DISPLAY" \
    -e "QT_X11_NO_MITSHM=1" \
    -v "/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    -e "XAUTHORITY=$XAUTH" \
    -e ROS_IP=127.0.0.1 \
    -e NVIDIA_DRIVER_CAPABILITIES=all \
    --cap-add=SYS_PTRACE \
    ov_ros2_20_04 bash
```

Altered the files referenced in 
_Originally posted by @ZibYang in https://github.com/rpng/open_vins/issues/316#issuecomment-1470342837_
and rebuilt the opencv files as mentoned in the install guide

```
git clone https://github.com/opencv/opencv/
git clone https://github.com/opencv/opencv_contrib/
mkdir opencv/build/
cd opencv/build/
cmake -DOPENCV_EXTRA_MODULES_PATH=../../opencv_contrib/modules ..
make -j8
sudo make install`
```

I'm able to colcon build without errors now but trying to launch gives me this

ros2 launch ov_msckf subscribe.launch.py config:=euroc_mav
Package 'ov_msckf' not found: "package 'ov_msckf' not found, searching: ['/opt/ros/galactic']"

I've sourced both /opt/ros/galactic/setup.bash and install/setup.bash

closed 2023-06-20 by nicko369

Aishkrish18, 2023-07-27:
[image]

Hey, 
I am still facing the same error as mentioned above. Although, on trying to build opencv from source, it still gives the same error. This is the same case for both ROS2 dockers since I tried both of them.  Could you suggest a solution for this if it has already worked out for any of you? Would be very much useful to have a ROS2 docker, since ros1 dockers works perfectly fine for me.

Regards,
AIshwarya