rpng/open_vins #349: Drone shoots away in the X-Y plane when doing vertical takeoff

meepo20, 2023-06-22:
The drone initially starts on the ground and doesn't move for a few seconds
Then a performs vertical takeoff.
The path should reflect the change in the Z axis however it just shoots off into the distance on the X-Y axis.
I've changed the parameters in the files to accommodate my drone

IMU parameters have been set to 0 
Camera has been calibrated according to messaged from /camera_info
[Image of path and take off height of drone](https://imgur.com/a/W377PWG)
https://imgur.com/a/W377PWG

nicko369, 2023-06-22:
https://drive.google.com/drive/folders/1id5e6_6XDSqAFTeTynhyxp-74kaDTIrC?usp=sharing
bag file and imu imucam configs used

meepo20, 2023-06-22:
ROS2 bag, IMU, IMU CAM config files 
https://drive.google.com/drive/folders/1id5e6_6XDSqAFTeTynhyxp-74kaDTIrC?usp=sharing

nicko369, 2023-06-22:
Forgot to add that his is a gazebo simulation

goldbattle, 2023-06-25:
The estimator needs to still have noise values even if the simulated measurements have no noise. I recommend using 1pixel and the noises from the [euroc_mav](https://github.com/rpng/open_vins/blob/master/config/euroc_mav/kalibr_imu_chain.yaml#L8C27-L12C93).
```
  accelerometer_noise_density: 2.0000e-3  # [ m / s^2 / sqrt(Hz) ]   ( accel "white noise" )
  accelerometer_random_walk: 3.0000e-3    # [ m / s^3 / sqrt(Hz) ].  ( accel bias diffusion )
  gyroscope_noise_density: 1.6968e-04     # [ rad / s / sqrt(Hz) ]   ( gyro "white noise" )
  gyroscope_random_walk: 1.9393e-05       # [ rad / s^2 / sqrt(Hz) ] ( gyro bias diffusion )
```

meepo20, 2023-06-25:
Hi Patrick. 
I was echoing the IMU data and saw that there  were still fluctuations in the readings and it wasn't exactly 0 when level.
I've performed the IMU calibration and obtained the values
 accelerometer_noise_density: 0.007205075424279859  # [ m / s^2 / sqrt(Hz) ]   ( accel "white noise" )
  accelerometer_random_walk: 3.122055276713202e-05    # [ m / s^3 / sqrt(Hz) ].  ( accel bias diffusion )
  gyroscope_noise_density: 0.0010819891643888133   # [ rad / s / sqrt(Hz) ]   ( gyro "white noise" )
  gyroscope_random_walk: 5.910826591997863e-06     # [ rad / s^2 / sqrt(Hz) ] ( gyro bias diffusion )

However the same issue still occurs.

goldbattle, 2023-07-15:
I still think this is likely a calibration problem with either the noises or transformation. Not sure if there is much we can do to help here. Feel free to reopen if you were able to solve it, or have any further progress.

closed 2023-07-15 by goldbattle