rpng/open_vins #350: Estimation diverges on Xavier NX with EuRoc

mzahana, 2023-06-25:
Hi

I have Jetson Xavier NX with a ROS2 `hubmle` docker container. I cloned this repo and built it using `colcon build`.
the build took some time, but it was successful.

I downloaded   one of the EuROC datasets , `V1_01_easy`. Then, I ran the estimator using `ros2 launch ov_msckf subscribe.launch.py config:=euroc_mav`. Then, I played the bag using `ros2 bag play V1_01_easy`.

I visulaized the `odomimu` in RViz2, and I see the estimator diverges quickly.

The same process works fine on my desktop PC.

What could be the issue? Xavier NX is not powerful enough ?

@goldbattle any hints?

Thanks in advance.

goldbattle, 2023-06-25:
I am not sure, and it is difficult for me to debug for you. I can recommend increasing the verbosity and seeing if there are any issues, and also inspecting the feature tracking by looking at the published images. Do you see that the estimator was able to successfully estimate, are features being extracted, and is the system publishing at full rate (use ros2 topic hz I believe to see this)?

I know that on the Jetson Nano with 5W mode the system struggles / fails, but on non-low power modes code works as expected. You also might see performance issues by running it inside docker, so you might want to run it directly on the system itself. Hope this helps.

mzahana, 2023-06-25:
Thanks @goldbattle  for the feedback. I am not sure if this helps, but here is some screen output
```
run_subscribe_msckf-1] subscribing to IMU: /imu0
[run_subscribe_msckf-1] subscribing to cam (stereo): /cam0/image_raw
[run_subscribe_msckf-1] subscribing to cam (stereo): /cam1/image_raw
[run_subscribe_msckf-1] [TIME]: 0.0581 seconds total (17.2 hz, 2.50 ms behind)
[run_subscribe_msckf-1] [TIME]: 0.0488 seconds total (20.5 hz, 2.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,0 < 15
[run_subscribe_msckf-1] [TIME]: 0.0367 seconds total (27.2 hz, 2.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,0 < 15
[run_subscribe_msckf-1] [TIME]: 0.0268 seconds total (37.4 hz, 1.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0145 seconds total (69.0 hz, 1.00 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0380 seconds total (26.3 hz, 2.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0189 seconds total (53.0 hz, 2.00 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0111 seconds total (89.9 hz, 3.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0376 seconds total (26.6 hz, 1.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0379 seconds total (26.4 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0158 seconds total (63.4 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0126 seconds total (79.4 hz, 1.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0118 seconds total (84.7 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0146 seconds total (68.4 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0132 seconds total (75.8 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0094 seconds total (105.9 hz, 2.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0271 seconds total (36.9 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0238 seconds total (42.1 hz, 2.00 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0194 seconds total (51.5 hz, 2.00 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0113 seconds total (88.2 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0131 seconds total (76.1 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0107 seconds total (93.5 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0172 seconds total (58.3 hz, 1.00 ms behind)
[run_subscribe_msckf-1] [init]: not enough feats to compute disp: 0,81 < 15
[run_subscribe_msckf-1] [TIME]: 0.0108 seconds total (92.5 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.108,0.138 (10.00 thresh)
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0125 seconds total (80.1 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.225,0.057 (10.00 thresh)
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0263 seconds total (38.0 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.315,0.089 (10.00 thresh)
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0213 seconds total (47.0 hz, 3.00 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.374,0.143 (10.00 thresh)
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0232 seconds total (43.1 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.410,0.138 (10.00 thresh)
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0214 seconds total (46.7 hz, 0.50 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.458,0.199 (10.00 thresh)
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0107 seconds total (93.7 hz, 1.00 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.430,0.266 (10.00 thresh)
[... 28 lines cut ...]
[run_subscribe_msckf-1] [TIME]: 0.0089 seconds total (111.9 hz, 5.00 ms behind)
[run_subscribe_msckf-1] [init]: disparity is 0.334,0.294 (10.00 thresh)
[... cut ...]
[run_subscribe_msckf-1] [init]: failed static init: no accel jerk detected
[run_subscribe_msckf-1] [TIME]: 0.0138 seconds total (72.6 hz, 4.00 ms behind)
```
* I run the Xaviern NX at the max MODE with 20W and 6 CORES
* The frequency of topic `/ov_msckf/trackhist` is around 6Hz
* The frequency of `/ov_msckf/odomimu` is at max of ~120Hz
* The frequency of `/ov_msckf/poseimu` is ~ 20Hz

I can record a ros 2 bag for debugging. What is the minimum set of topics to record ?

Thanks again for your assistance.

mzahana, 2023-06-25:
[Here](https://drive.google.com/file/d/1JEuEsR7l_uE1F1y356Pszq-VRoQCI1kK/view?usp=sharing) is a ros 2 bag

goldbattle, 2023-06-25:
What you have printed looks ok. Can you try the `DEBUG` verbosity level? Publishing of trackhist seems to be a bit slow. Are you recording this over a network? If so, check the `/compressed` topic instead as sending the full image over will give lower frequency / not full rate.

Looking at `ros2 topic hz /ov_msckf/poseimu -w 100` it seems things are only being processed at 10Hz, so likely the feature tracking or something is too slow on the platform. You can try just running 1 camera, or reducing the number of features to see if this provides any insight.

mzahana, 2023-06-25:
Thank you very much @goldbattle  for taking the time to check my ros bag.

I recorded the bag on the same platform, not over a network.

If it worked OK with you on a jetson nano, then it makes sense that it should work on Xavier NX as it has more compute power. Also, I doubt that running the system inside docker is the main bottleneck. I have used more complex software inside docker containers with no issues.

I think  feature tracking has some issues as it seems to be running at low speed, but I am not sure why. I will try to play with the number of features and number of cameras. 

What number of features did you use on the Nano?

mzahana, 2023-06-26:
I set `verbosity:=DEBUG` and I get the following screen output
```
[run_subscribe_msckf-1] ROS2Visualizer.cpp:49 Publishing: /ov_msckf/poseimu
[run_subscribe_msckf-1] ROS2Visualizer.cpp:51 Publishing: /ov_msckf/odomimu
[run_subscribe_msckf-1] ROS2Visualizer.cpp:53 Publishing: /ov_msckf/pathimu
[run_subscribe_msckf-1] ROS2Visualizer.cpp:57 Publishing: /ov_msckf/points_msckf
[run_subscribe_msckf-1] ROS2Visualizer.cpp:59 Publishing: /ov_msckf/points_msckf
[run_subscribe_msckf-1] ROS2Visualizer.cpp:61 Publishing: /ov_msckf/points_aruco
[run_subscribe_msckf-1] ROS2Visualizer.cpp:63 Publishing: /ov_msckf/points_sim
[run_subscribe_msckf-1] ROS2Visualizer.cpp:67 Publishing: /ov_msckf/trackhist
[run_subscribe_msckf-1] ROS2Visualizer.cpp:71 Publishing: /ov_msckf/posegt
[run_subscribe_msckf-1] ROS2Visualizer.cpp:73 Publishing: /ov_msckf/pathgt
[run_subscribe_msckf-1] ROS2Visualizer.cpp:175 subscribing to IMU: /imu0
[run_subscribe_msckf-1] ROS2Visualizer.cpp:200 subscribing to cam (stereo): /cam0/image_raw
[run_subscribe_msckf-1] ROS2Visualizer.cpp:201 subscribing to cam (stereo): /cam1/image_raw
[run_subscribe_msckf-1] run_subscribe_msckf.cpp:102 done...spinning to ros
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0394 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0411 seconds total (24.3 hz, 10.00 ms behind)
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0000 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0250 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0253 seconds total (39.6 hz, 5.00 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,0 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0002 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0339 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0342 seconds total (29.3 hz, 7.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,0 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0003 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0212 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0215 seconds total (46.6 hz, 2.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0003 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0174 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0176 seconds total (56.9 hz, 2.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0003 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0112 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0114 seconds total (87.4 hz, 3.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0003 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0202 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0205 seconds total (48.9 hz, 5.00 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0003 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0147 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0149 seconds total (67.0 hz, 7.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0003 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0116 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0119 seconds total (84.3 hz, 2.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0002 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0184 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0186 seconds total (53.7 hz, 0.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0004 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0166 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0169 seconds total (59.3 hz, 2.50 ms behind)
[run_subscribe_msckf-1] InertialInitializer.cpp:117 [init]: not enough feats to compute disp: 0,76 < 15
[run_subscribe_msckf-1] VioManagerHelper.cpp:170 [init]: failed initialization in 0.0006 seconds
[run_subscribe_msckf-1] VioManager.cpp:314 [TIME]: 0.0178 seconds for tracking
[run_subscribe_msckf-1] ROS2Visualizer.cpp:483 [TIME]: 0.0180 seconds total (55.5 hz, 0.50 ms behind)
[... cut ...]
[run_subscribe_msckf-1] VioManager.cpp:615 [TIME]: 0.0317 seconds for MSCKF update (8 feats)
[run_subscribe_msckf-1] VioManager.cpp:617 [TIME]: 0.0100 seconds for SLAM update (47 feats)
[run_subscribe_msckf-1] VioManager.cpp:618 [TIME]: 0.0184 seconds for SLAM delayed init (4 feats)
[run_subscribe_msckf-1] VioManager.cpp:620 [TIME]: 0.0062 seconds for re-tri & marg (11 clones in state)
```

* I set the number of features is 150/100/70, and it still diverges
* I set `use_stereo:=false` and `max_cameras:=1`, but it also diverges

mzahana, 2023-06-26:
Here is the output of  `ros2 run ov_core test_profile`
```
Setting printing level to: INFO
OPENCV: RANDOM BIG IMAGE: 21.280 +- 8.770ms
OPENCV: HISTOGRAM EQUALIZATION: 8.874 +- 7.351ms
OPENCV: BUILD OPTICAL FLOW PYRAMID: 5.000 +- 5.773ms
OPENCV: FAST FEATURE EXTRACTION: 33.970 +- 8.910 ms (90402.12 avg feats)
OPENCV: KLT OPTICAL FLOW: 18.359 +- 3.239 ms (415.51 avg feats)
EIGEN3(double): RANDOM BIG MATRIX: 6.252 +- 0.983ms
EIGEN3(double): MATRIX MULTIPLY: 9.013 +- 1.296ms
EIGEN3(double): MATRIX INVERSION: 17.601 +- 1.774ms
EIGEN3(double): HOUSEHOLDER QR FULL: 36.662 +- 2.900ms
EIGEN3(double): HOUSEHOLDER QR PIV: 44.418 +- 2.914ms
EIGEN3(double): HOUSEHOLDER QR CUSTOM: 43.052 +- 2.203ms
EIGEN3(float): RANDOM BIG MATRIX: 4.720 +- 0.796ms
EIGEN3(float): MATRIX MULTIPLY: 4.409 +- 0.576ms
EIGEN3(float): MATRIX INVERSION: 11.001 +- 1.417ms
EIGEN3(float): HOUSEHOLDER QR FULL: 167.093 +- 5.698ms
EIGEN3(float): HOUSEHOLDER QR PIV: 24.115 +- 1.516ms
EIGEN3(float): HOUSEHOLDER QR CUSTOM: 22.997 +- 1.670ms
```

HoEmpire, 2023-06-26:
We were able to run openvins with a T265 on Jetson NANO in real-time with the following important changes in the default config of EurocMAV:
```txt
max_slam: 25
max_slam_in_update: 25
max_msckf_in_update: 20
num_pts: 50
downsample_cameras: true #this reduces the tracking time and CPU workload by a lot
```

The runtime statistics are
```txt
 mean_time = 0.0192 | std = 0.0084 | 99th = 0.0388  | max = 0.0603 (tracking)
mean_time = 0.0006 | std = 0.0002 | 99th = 0.0012  | max = 0.0092 (propagation)
mean_time = 0.0008 | std = 0.0013 | 99th = 0.0039  | max = 0.0217 (msckf update)
mean_time = 0.0030 | std = 0.0022 | 99th = 0.0082  | max = 0.0200 (slam update)
mean_time = 0.0023 | std = 0.0085 | 99th = 0.0222  | max = 0.0915 (slam delayed)
mean_time = 0.0003 | std = 0.0002 | 99th = 0.0008  | max = 0.0088 (re-tri & marg)
mean_time = 0.0263 | std = 0.0121 | 99th = 0.0543  | max = 0.1129 (total)
```

mzahana, 2023-06-27:
@HoEmpire I tried your configs, but the estimation drifts a lot.
Can you share how you setup open_vins on Jetons Nano?
Maybe I am missin something related to the setup of the dependencies such as opencv, Eigen, Ceres

HoEmpire, 2023-06-27:
The Jetson Nano is not with me at this moment, so I am not sure what versions the opencv and Eigen are.
We only tested this config live running openvins with a real sense T265, the trajectory looks reasonably good, but we didn't evaluate the accuracy. And we didn't test this configuration on EuRoc dataset at that time.
I just quickly checked this config on my **laptop** (opencv 4.2.0, eigen 3.3.7, ceres should not be a problem if not using dynamic initialization if I remember correctly) and got the following result
```txt
============================================
ATE LATEX TABLE
============================================
 & \textbf{V1\_01\_easy} & \textbf{V1\_02\_medium} & \textbf{V1\_03\_difficult} & \textbf{V2\_02\_medium}  \\\hline
openvins & 0.827 / 0.061 & 1.585 / 0.135 & 3.000 / 0.142 & 1.072 / 0.120  \\
============================================
```
It failed on V201 and V203. But at least I assume this config should be able to run on V101 if the tracking performs normally. 
You can try uncommenting these two lines(https://github.com/rpng/open_vins/blob/master/ov_msckf/src/ros1_serial_msckf.cpp#L66-L67) to disable multi-threading and rebuild. Then ``roslaunch ov_msckf serial.launch``  to run things in serial to see if it helps.

WZG3661, 2023-06-29:
When I ran the euroc `MH_01_easy` dataset, it drifts far away quickly, but if I set the `downsample_cameras: ture`, the estimation was just OK.

goldbattle, 2023-06-29:
You should take a look at the script folder to see how we run all the
datasets. There are some non-static periods in the machine hall which
require addressing to initialize if I remember correctly.

mzahana, 2023-06-30:
@WZG3661 I set `downsample_cameras: ture`, but still the estimation diverges.

@goldbattle I saw the scripts folder. For the easy ros bag, I saw the start time is at 0, so there is no difference to my case as I start estimator  first and then play the bag.

I am not sure what is missing.

goldbattle, 2023-10-28:
One difference is that it is over the ROS communication instead of reading the bag directly, so this can cause some differences. There might be some issues with limited resources and the current multi-threading implementation, I am not sure. I know one problem is that the KLT tracking from OpenCV will have lag spikes which could also cause a problem. Unfortunately I can't debug much from here for you.

makeecat, 2023-12-07:
OpenVINS works fine with some tuning for ROS1. Once we switched to ROS2 and solved intra-process communication between the realsense camera and OpenVINS, the estimation diverge was worse than ROS1. The pose can float around when the stereo camera is almost static and stereo VIO can lose tracking very easily. 

I am also suspecting multi-threading implementation for ROS2 has some issues. To be clear, we have made our own composable node wrapper and compose both realsenseu node and openvins node into one container, and launched the container with the multi-threading mode.

ShreyasKuntnal, 2024-07-24:
@makeecat can you please specify what sort of tunes have you done in ros1 open vins implementation?
I am continuously facing drifting issues when we run openvins on d435i. And none of the issues in the GitHub specify some methodology to solve this.
So hoping to see ur tuning might give some magical improvement to my openvins implementation.