rpng/open_vins #351: Unable to perform state estimation for monocular cameras

Aishkrish18, 2023-06-26:
Hey @goldbattle , thanks for your reply in one of the issues. I am creating a new issue here now for the same problem . I have a short question. I am running d455 using Open vins in ros1 noetic and after this output shown below I don't see any tracking at all. I am not quite sure where I went wrong. I disabled stereo mode everywhere and changed to mono with 1 camera. I am trying to run the estimation live with my d455 camera .
You mentioned that it might be a problem with verbosity. Could you elaborate a bit about that here? I checked the topics from the camera that are published and they seem to be publishing fine at the rate mentioned in the steps.  Thank you so much for your help!
[image]

goldbattle, 2023-06-26:
Can try adding `verbosity:=DEBUG` to see what is happening.
https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/simulation.launch#L8C16-L8C25

Ensure you are using the latest code from the master branch.

goldbattle, 2023-07-19:
Feel free to reopen if you were unable to figure it out (with a bag too).

closed 2023-07-19 by goldbattle