rpng/open_vins #355: cmake broken for Windows

antithing, 2023-07-11:
Hi, I have the 2.7 release code , and see some errors when running cmake. 

There are no entries in the cmakelist.txt to create the projects / libs. 

For example, in the 1.0 release, you have:
```

# Include our header files
include_directories(
        src
        ${EIGEN3_INCLUDE_DIR}
        ${Boost_INCLUDE_DIRS}
        ${catkin_INCLUDE_DIRS}
)

# Set link libraries used by all binaries
list(APPEND thirdparty_libraries
        ${Boost_LIBRARIES}
        ${OpenCV_LIBRARIES}
        
)

##################################################
# Make the core library
##################################################
add_library(ov_core_lib
        src/dummy.cpp
        src/sim/BsplineSE3.cpp       
        src/track/TrackBase.cpp
        src/track/TrackAruco.cpp
        src/track/TrackDescriptor.cpp
        src/track/TrackKLT.cpp
        src/track/TrackSIM.cpp
        src/types/Landmark.cpp
        src/feat/Feature.cpp
        src/feat/FeatureInitializer.cpp
)
target_link_libraries(ov_core_lib ${thirdparty_libraries})

##################################################
# Make binary files!
##################################################
add_executable(test_tracking src/test_tracking.cpp)
target_link_libraries(test_tracking ov_core_lib ${thirdparty_libraries})

```

These entries are missing from 2.7, and so cmake doesn't work.

Would it be possible to share the working files so that I can compile on Windows, with no catkin or ROS?

Thanks!

antithing, 2023-07-13:
Understood, thank you. i have it compiling and can run the webcam test on windows. 

Now to pass in some actual data! Do you have any example code for running a dataset without using ROS?

goldbattle, 2023-07-15:
The only demo is running simulation. You could also checkout this small plugin I made for another project here:
https://github.com/goldbattle/ILLIXR/tree/upstream-OpenVINS/pose_openvins

closed 2023-07-19 by goldbattle