rpng/open_vins #357: Simple Tutorial RViz launch missing `-d` path to `.rviz` file

cod3monk3y, 2023-07-18:
Under the [ROS 1 section for the Simple Tutorial](https://docs.openvins.com/gs-tutorial.html#gs-tutorial-ros1), the launch command reads

```
rviz # term 2
```

Executing that directly will showed `Fixed Frame [map] does not exist` because RViz is using `default.rviz` configuration. I had to specify the `.rviz` file using `-d` to get the visualization working:

```
$ cd ~/openvs_ws/src/open_vins
$ rviz -d ov_msckf/launch/display.rviz
```

After this it works great!

Is this just my setup, or should the doc change from just `rviz` to something like: `rviz -d rviz -d ov_msckf/launch/display.rviz`?

cod3monk3y, 2023-07-19:
I assume yes, since you do this [in the demo video](https://youtu.be/rBT5O5TEOV4?t=2351).

goldbattle, 2023-07-19:
There is a comment above that about what config to open
> For RVIZ, one can open the ov_msckf/launch/display.rviz configuration file.

But directly from command line might help those newer to the project

PR #395 (2023-10-29): Small fixes
- fixes #357
- fixed #316 aruco building on opencv 4.7 and above
- fix #358 calibration file documentation for gyro rotation intrinsics
- added ubuntu 22.04 docker (seems ROS1 is dropped for ubuntu 22.04 forward :cry: )
commit f3ba270a (2023-10-29): fixes #357
closed 2023-10-29 by goldbattle