rpng/open_vins #360: Question about ZUPT derivation

tianlong312, 2023-07-19:
[image]

Hi there, I am reading the docs about ZUPT derivation from OpenVINS website and I have a question about the derivative (dz)/(db), should the sign be positive instead of negative in the derivation since z is actually -w_m+b_g+n_g?

goldbattle, 2023-07-20:
We take the derivative of the following function:
[image]

Which gives you the negative for all three Jacobians. There is then a discussion the page about how we compute the residual as the "true" measurement is actually us saying that we should have zero acceleration and zero angular velocity. You can see the implementation here:
https://github.com/rpng/open_vins/blob/e274c656324f84b9f2730eacd0cf54a6f8b74bb8/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L160-L178

tianlong312, 2023-07-20:
Thanks for your reply, does that mean the notation from the OpenVINS website is wrong? Instead of taking derivative of the residual as in the documentation, we are taking derivative of the “true ”measurement.

goldbattle, 2023-07-23:
For EKF / filter system we always formulate the function $$z_m = h(x) + n$$ where we then find the derivative in respect to. There is some subtilty with having the error state for orientation. This function is then re-arranged to give $$\tilde{z} = z_m - h(\hat{x}) = \mathbf{H} \tilde{x} + n$$
Hope this helps.

closed 2023-08-03 by goldbattle