rpng/open_vins #361: Can the init phase be skipped?

cod3monk3y, 2023-07-20:
Thanks for [explaining `init_window_time` and `init_imu_thresh`](https://github.com/rpng/open_vins/issues/359). I'm able get most new videos I record (and convert to .bag using gopro_ros) to track properly by doing an initial rest of ~0.5-1.0 and tuning these properties accordingly.

However, I have some existing videos I've recorded when I was not aware of this init period requirement.

1. Is it possible to skip `init`?
2. What is happening during this phase? Noise fine tuning? Establishing initial frame of reference? 

I've attempted to bypass init by re-writing the bag file and duplicating the first message on each topic back in time N seconds (re-writing the timestamps in both the bag file and the message headers; accounting for the sample rates by comparing the deltas for each topic). I've verified this _does sort of work_ to trick the init/jerk detection. But I fear this could be stomping on some other calculations happening during init.

Here's an example of a modified .bag file with the extra data injected at the front:

[image]

How terrible of an idea is this?

cod3monk3y, 2023-07-20:
Ah, Sorry. I think I'm looking for [dynamic initialization](https://docs.openvins.com/classov__init_1_1DynamicInitializer.html). Sorry for not reading the docs.

cod3monk3y, 2023-07-21:
Modifying the question now that I realize I need Dynamic Init: 

I have a workflow now that works well with static initialization, but I'm not able to get **dynamic init** working. The init output looks like this (showing transitions only):

```
[TIME]: 0.0023 seconds total (434.8 hz, 1.05 ms behind)
[init]: not enough feats to compute disp: 0,60 < 15
...
[TIME]: 0.0025 seconds total (404.5 hz, 1.03 ms behind)
[init]: disparity is 2.692,48.366 (2.00 thresh)
[TIME]: 0.0027 seconds total (375.0 hz, 1.01 ms behind)
[init]: disparity is 4.463,50.230 (2.00 thresh)
...
```

and shortly dies in `init-d`

```
[init]: disparity is 50.670,53.546 (2.00 thresh)
[TIME]: 0.0036 seconds total (280.5 hz, 1.36 ms behind)

[init-d]: velocity in I0 was 0.126,0.328,-0.088 and |v| = 0.3620
[init-d]: gravity in I0 was -0.265,0.951,9.760 and |g| = 9.8100
[init-d]: 35 iterations | 6 states, 32 feats (32 valid) | 207 param and 413 res | cost 4.6742e+02 => 1.1983e+02
================================================================================REQUIRED process [ov_msckf-2] has died!
```

Is there a guide for setting the `init_dyn*` parameters? I've basically left everything alone except for flipping `init_dyn_use` to `true`.

```
init_dyn_use: true # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] # initial gyroscope bias guess
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] # initial accelerometer bias guess
```

cod3monk3y, 2023-07-21:
Here is another run final output before the process dies.

```
[TIME]: 0.0058 seconds total (172.3 hz, 1.43 ms behind)
[init]: disparity is 114.872,31.195 (2.00 thresh)

[init-d]: velocity in I0 was 0.146,0.214,-0.116 and |v| = 0.2840
[init-d]: gravity in I0 was -0.850,4.683,8.578 and |g| = 9.8100
[init-d]: not enough features for our mle (1 < 8)!

[TIME]: 0.0074 seconds total (134.4 hz, 1.22 ms behind)
[init]: disparity is 95.136,29.545 (2.00 thresh)

[init-d]: velocity in I0 was 0.170,0.149,0.006 and |v| = 0.2262
[init-d]: gravity in I0 was -0.907,4.452,8.694 and |g| = 9.8100
[init-d]: 43 iterations | 6 states, 8 feats (8 valid) | 135 param and 155 res | cost 1.1399e+02 => 1.0483e+02
```

cod3monk3y, 2023-07-21:
Here's another run's output before `ov_msckf` dies

```
[TIME]: 0.0024 seconds total (417.5 hz, 1.25 ms behind)
[init]: disparity is 56.116,52.103 (2.00 thresh)

[TIME]: 0.0022 seconds total (461.3 hz, 1.23 ms behind)
[init-d]: velocity in I0 was 0.140,0.242,-0.068 and |v| = 0.2872
[init-d]: gravity in I0 was -0.258,0.959,9.760 and |g| = 9.8100
[init-d]: 43 iterations | 7 states, 37 feats (37 valid) | 238 param and 482 res | cost 5.3711e+02 => 1.5564e+02
=== ...
process has died

```

goldbattle, 2023-09-06:
Please post a bag that is able to reproduce this failure. From the message, I am not sure what is wrong. You are using the latest release correct?

goldbattle, 2023-09-14:
Feel free to reopen with the data if you still have trouble. You can also try running on another platform to see if it is related to the specific ceres install on that computer.

closed 2023-09-14 by goldbattle