rpng/open_vins #362: parameters from dynamic initialize not used

biggiantpigeon, 2023-07-21:
I notice that parameters from dynamic initialize not used, as in code
`else if (params.init_dyn_use) {
    PRINT_DEBUG(GREEN "[init]: USING DYNAMIC INITIALIZER METHOD!\n" RESET);
    std::map<double, std::shared_ptr<ov_type::PoseJPL>> _clones_IMU;
    std::unordered_map<size_t, std::shared_ptr<ov_type::Landmark>> _features_SLAM;
    return init_dynamic->initialize(timestamp, covariance, order, t_imu, _clones_IMU, _features_SLAM);
  }`
this means all poses from the time before init success is cleared in sliding window(and also the features in database), and we rebuild the sliding windows afterwards. Why don't we use these poses and features? Any concern? As I recall, neither orbslam3 or VINS-mono don't discard them. Thanks!

goldbattle, 2023-08-03:
You are asking why we don't directly initialize the sliding window and features in addition to the inertial 15dof state. This is mainly because recovering the initial covariance of the whole system is expensive, so this will likely be a bottleneck. I think I have some TODOs in the code as it would be nice to recover the features and directly use them. The system should be more stable in this case, but not sure.

biggiantpigeon, 2023-08-18:
Oh I see, thanks a lot~

closed 2023-08-18 by biggiantpigeon