rpng/open_vins #365: Found Negative at StateHelper::EKFUpdate() - diagonal

jiratto, 2023-07-31:
My hardware configuration
- stereo camera
- imu from pixhwak via Mavros
- SOC ARM Amlogic A311D
- ubuntu 20.04 with ROS1
- opencv 4.6

when I run full system, "I found StateHelper::EKFUpdate() - diagonal is Negative "on run time. But when I record bag with this configuration and run on my PC, there is no problem. 

my open_vins configuration
`
verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true
integration: "rk4"
use_stereo: true
max_cameras: 2

calib_cam_extrinsics: false
calib_cam_intrinsics: false
calib_cam_timeoffset: false
calib_imu_intrinsics: false
calib_imu_g_sensitivity: false

max_clones: 11
max_slam: 50
max_slam_in_update: 25
max_msckf_in_update: 40
dt_slam_delay: 1

gravity_mag: 9.81

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

try_zupt: false
zupt_chi2_multipler: 0
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5
zupt_only_at_beginning: false

init_window_time: 2.0 
init_imu_thresh: 3.5 
init_max_disparity: 15.0 
init_max_features: 75 

init_dyn_use: true 
init_dyn_mle_opt_calib: false 
init_dyn_mle_max_iter: 100
init_dyn_mle_max_time: 0.10 
init_dyn_mle_max_threads: 6 
init_dyn_num_pose: 6 
init_dyn_min_deg: 10.0
init_dyn_inflation_ori: 10
init_dyn_inflation_vel: 100
init_dyn_inflation_bg: 10
init_dyn_inflation_ba: 100
init_dyn_min_rec_cond: 1e-12 

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] 
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] 

record_timing_information: false 
record_timing_filepath: "/tmp/traj_timing.txt"

save_total_state: false
filepath_est: "/tmp/ov_estimate.txt"
filepath_std: "/tmp/ov_estimate_std.txt"
filepath_gt: "/tmp/ov_groundtruth.txt"

use_klt: true 
num_pts: 200 
fast_threshold: 20 
grid_x: 5 
grid_y: 5 
min_px_dist: 10
knn_ratio: 0.70 
track_frequency: 21.0
downsample_cameras: false 
num_opencv_threads: 4 
histogram_method: "HISTOGRAM" 

use_aruco: false
num_aruco: 1024
downsize_aruco: true

up_msckf_sigma_px: 1
up_msckf_chi2_multipler: 1
up_slam_sigma_px: 1
up_slam_chi2_multipler: 1
up_aruco_sigma_px: 1
up_aruco_chi2_multipler: 1

use_mask: false

relative_config_imu: "2023-07-21-08-03-45-imu.yaml"
relative_config_imucam: "2023-07-21-08-03-45-camchain-imucam.yaml"
` 
[image]

[image]

goldbattle, 2023-08-03:
I recommend increasing the verbosity and seeing if any errors are being reported. An update time of 0.19 seconds before this happening is a bit suspicious. Most of the time these are caused by poor calibration or noise parameters, but it is hard to tell.

jiratto, 2023-08-03:
@goldbattle Thank you for your advice
I changed verbosity and this is result.
[image]

jiratto, 2023-08-03:
[image]

goldbattle, 2023-08-23:
Were you able to debug this problem at all? Did you try reducing the number of tracked features and SLAM features in the state to see if was related to computational load problems?

closed 2023-09-14 by goldbattle