rpng/open_vins #366: State estimation completely goes out of the rviz grid on performing with ZED2

Aishkrish18, 2023-07-31:
Hey,
First of all thank you so much for this great open source work and for your instant replies for clearing our doubts. 
I am performing state estimation for ZED2 camera and I am currently facing this problem where the tracking goes off the rviz grid and it shoots off. I perfomed correct calibration and i have pasted them here. I am not sure if there are certain specific parameters that i have to tune for having them exactly follw the path of the camera. 

I have herewith attavched the calibration files and the screenshot of how it shoots off from the grid. 
[image]
[image]
[image]
[image]
[zed2_cam_imu_calib4-results-imucam.txt](https://github.com/rpng/open_vins/files/12212942/zed2_cam_imu_calib4-results-imucam.txt)

Aishkrish18, 2023-08-01:
Update: This action was somewhat similar for d435 as well. It would be great if someone can assist me on how to fix it, so that i can compare the results and make a conclusion.  i plan to use this algorithm for my master thesis to work and tune more on top of the existing algorithm :)

Aishkrish18, 2023-08-01:
By imu intrinsics you mean the allan variance ros -imu  bias values or imu-camera calibration?  Or am i missing something other than that?

goldbattle, 2023-08-01:
Kalibr IMU-camera calibration, should be the --imu-models parameter to
scale-missalignment I believe?
Allan variance will get your noise parameters, not the intrinsics.

Aishkrish18, 2023-08-02:
Not sure if i understood you correctly,  but you mean I should calibrate imu camera using the model parameter scale-misalignment instead of "calibrated" is it? for ZED2 and d435? This will help solving the issue?

Aishkrish18, 2023-08-02:
[image]
[image]

On performing calibration once again by using scale-misalignment, i was able to retrieve the intrinsics of IMU. But this is still not helping and it still continues to shoot out of the grid. I have herewith attached the values  and the cam chain files 
[zed2_cam_imu_calib4-results-imucam.txt](https://github.com/rpng/open_vins/files/12240838/zed2_cam_imu_calib4-results-imucam.txt)
[zed2_cam_imu_calib4-report-imucam.pdf](https://github.com/rpng/open_vins/files/12240839/zed2_cam_imu_calib4-report-imucam.pdf)

yangyulin, 2023-08-02:
Have you fully excited the motion when doing calculations?It is clear that the IMU is not good, The scale correction for gyro and accel is too largeSent from my iPhone

goldbattle, 2023-08-03:
There are also some discussions about using ZED / ZED2 if you search the issues:
- https://github.com/rpng/open_vins/issues/52
- https://github.com/rpng/open_vins/issues/133
- https://github.com/rpng/open_vins/issues/157

goldbattle, 2023-08-23:
Also feel free to pose your ZED2 configs if you are able as it seems quite a few people have tried to use this device. Many thanks!

closed 2023-08-23 by goldbattle