rpng/open_vins #367: Loss of tracking stability after > 15 seconds

cod3monk3y, 2023-08-02:
Using gopro_ros conversion of Hero9 video to .bag files, I've been able to get some very good tracking in OpenVINS (thanks for the great work on this project!). _However_, sometimes the tracking becomes unstable *even after 10s of seconds of stability* and "shoots off into nowhere" like in [Issue 366](https://github.com/rpng/open_vins/issues/366). I dug into this a bit, but I'm wondering if you have any feedback to help me tune this so I can maintain stable tracking throughout. (Or, maybe this is a bug?)

To establish and determine stable tracking, I follow these tips:

1. use a stable starting position on a fixed surface, for 0.5-1.0 seconds
2. ensure there are many visible features during (1), e.g. staring at a blank wall will cause failure to init (_not that I would do something so ridiculous..._)
3. use a "jerk" motion when lifting in order to pass the "init" accelerometer variance check, otherwise tracking won't start. If init still fails, check accelerometer timing using PlotJuggler and tune config file accordingly. 
4. set proper values in config file for (1) and (3)
5. walk / drive a loop
6. be "cooperative" with the device to ensure there is no _very fast rotation_ and that there _are always features in the video_
7. make sure lighting does not rapidly change (e.g. 30 lux to 2000 lux) since this seems to cause a large scale loss / reevaluation of visible features
9. return to the original starting position
10. measure distance between start and end position (use custom script to capture distance from starting point)

With these tips, I'm able to get very good tracking. Here is an example of good (great), stable tracking along with the count of msckf points.

[image]

I've identified something I think may be causing the tracking to become unstable.

The following plot shows the speed of the camera (calculated from the vio position topic), and the number of features on topics `/ov_msckf/points_msckf` and `/ov_msckf/loop_feats`. There is a period from 18.8-20.1 seconds where there are zero features in `loop_feats`. This is when tracking loses stability and does not recover.

[image]

Speed axis is limited to 4 m/s as the maximum human speed, and exceeding this value is a signal that tracking is lost.

Here is another example, where tracking becomes unstable after 70 seconds. I do not have point cloud counts for this one.

[image]

Questions

1. Should I expect OpenVINS to have stable tracking even when there are no visible features tracked for several consecutive frames. I.e. shouldn't the IMU handle these brief periods without becoming unstable?
2. Are there parameters I can tune re: (1)
3. I'm running ov_msckf at the default 15Hz. Would increasing this to 30Hz help? 
4. Any other thoughts on why this might be becoming unstable?

Hardware setup: GoPro, 1080p Wide (down-scaled 0.5x), 30fps, 200Hz IMU, all stabilization disabled, fully calibrated using Kalibr.

Thanks again for your hard work!

goldbattle, 2023-08-03:
I would double check that you have good IMU during this time. It depends on the quality of the IMU used, if the biases of the system have been accurately estimated, and if the noise values are correct. But yes, it should be able to handle maybe 1 second or so of bad feature tracking but not something we have extensively tested (you could try blacking out the images on the eurocmav for a few seconds throughout the dataset to see how it does there with a good calibration dataset).

cod3monk3y, 2023-08-03:
Thanks for the quick response. The Kalibr/Allan Variance reports and graphs all looked very good. I followed the noise/random walk multipliers from the config files. 

Do you suggest increasing or decreasing the multiplier as a test for the noise/walk scaling?

I will try blacking out the eurocmav as a test. Thanks for the idea.

I will also try running the Allan Variance again to see if I get consistent noise values.

goldbattle, 2023-08-03:
Can you post your full kalibr pdf report you have generated?
I was thinking the timesteps or IMU intrinsics are poor possibly, but not sure.
During this time, maybe there as bad measurements or something (could look via rqt_bag if you have the bag still).

goldbattle, 2023-09-14:
Feel free to reopen with the data if you still have trouble.

closed 2023-09-14 by goldbattle