rpng/open_vins #370: Doesn't work on processed rosbags

Jominsu99, 2023-08-16:
Hello i need help.
The raw rosbag file works fine for both serial.launch and subscribe.launch. However, when using processed rosbag topics, the following error occurs.
I brought an example, but as a result of converting the compressed image into a raw image and executing serial.launch, it does not work with the following error message.

[image]

What could be the problem?

processed bag topic raw -> compressed
t.bag is part of handheld.bag.

[image]

estimator_config.yaml
kalibr_imucam_chain.yaml

[image]

kalibr_imu_chain.yaml

[image]

Jominsu99, 2023-08-17:
Apart from the problem above, the same error as above occurred when serial.launch was executed with a data set that did not have an error with subscribe.launch.

[image]

I don't know how to fix it. I need help

goldbattle, 2023-08-23:
It looks like the dataset has finished being processed correctly. In your first screenshot where no information is posted likely means the topic names are incorrectly configured or the timestamps of the converged images are no longer correct. I recommend investigating this and inspecting with `rqt_bag` utility.

closed 2023-09-06 by goldbattle