rpng/open_vins #372: One possible bug about Qc

KleistvonLiu, 2023-08-21:
I think I found a bug from line 459 of Propagator.cpp:

// Construct our discrete noise covariance matrix
// Note that we need to convert our continuous time noises to discrete
// Equations (129) amd (130) of Trawny tech report
Eigen::Matrix<double, 12, 12> Qc = Eigen::Matrix<double, 12, 12>::Zero();
Qc.block(0, 0, 3, 3) = std::pow(_noises.sigma_w, 2) / dt * Eigen::Matrix3d::Identity();
Qc.block(3, 3, 3, 3) = std::pow(_noises.sigma_a, 2) / dt * Eigen::Matrix3d::Identity();
Qc.block(6, 6, 3, 3) = std::pow(_noises.sigma_wb, 2) / dt * Eigen::Matrix3d::Identity();
Qc.block(9, 9, 3, 3) = std::pow(_noises.sigma_ab, 2) / dt * Eigen::Matrix3d::Identity();

Should it be:
Qc.block(6, 6, 3, 3) = std::pow(_noises.sigma_wb, 2) * dt * Eigen::Matrix3d::Identity();
Qc.block(9, 9, 3, 3) = std::pow(_noises.sigma_ab, 2) * dt * Eigen::Matrix3d::Identity();

goldbattle, 2023-08-21:
Hi, have you compared against the technical report and searched all issues?
I remember this being discussed many times before, so please search first
and feel free to link to the issues if that doesn't answer your question.

Additionally, please provide a bit of reasoning why it "should be" what you
have posted.
Thanks!

KleistvonLiu, 2023-08-21:
I get it! Thanks for your reply!

closed 2023-08-21 by KleistvonLiu