rpng/open_vins #373: Failed to initialize: D455+Jetson Orin on a drone

mzahana, 2023-08-22:
Hi @goldbattle 

I have the following setup
* Jetson Orin NX
* open_vins `master` inside a docker image
* Realsense D455
The above is mounted on a drone

failed static init: no accel jerk detected

goldbattle, 2023-08-23:
I recommend directly trying the d455 config:
https://github.com/rpng/open_vins/tree/master/config/rs_d455

The `init_window_time` shouldn't be more than 2 seconds, and you would just need to tune the `init_imu_thresh` to detect your device pickup if needed. This above config should work out of the box so first try this / compare your configuration to it.

mzahana, 2023-08-23:
Thank you very much @goldbattle . I will try it and report the result.

Could you please elaborate more on `init_imu_thresh` parameter ?
How is it used ?
Thanks.

goldbattle, 2023-08-26:
You can see that we compute the accelerometer standard deviations. If this is large then we expect the platform to have changed its acceleration or rotation. We have two windows. The first window we want to be static, and the second to have this high variance in acceleration (e.g. it was picked up). We use the first window to recover the direction of gravity to initialize the system. This is the logic for static initialization.
https://github.com/rpng/open_vins/blob/master/ov_init/src/static/StaticInitializer.cpp#L84-L98

`init_imu_thresh` will depend on how noisy the IMU is. For example if you have a MAV with motors this will cause noise and thus will require this threshold to be increased to prevent the case that turning the motors on to introduce large acceleration noise from the vibrations. It also depends on how noisy the IMU is.

In general this shouldn't be too hard to tune, but seems there are quite a few issues about how to do this. I recommend recording a bag of picking up the sensor, and then based on this tune the parameter such that when it is picked up (inspect the camera feed) the system detects it as moving. You want this window to be short (between 1-1.5 seconds) so you have good responsiveness. 

Hope this helps.

Userpc1010, 2023-08-30:
@mzahana If you run open vins successfully, could you make a short demo of how tracking works on the drone?

mzahana, 2023-08-30:
Thanks @goldbattle for your feedback. I am doing some experiments and will share some feedback soon.

@Userpc1010 I think it can work pretty well, depending on your environment, camera-imu setup, calibration, and compute power. I am still doing some testing.

mzahana, 2023-09-03:
I was finally able to run `ov_msckf` successfully on a quadcopter. In my setup, I use D455 connected to a Jetson Orin NX. I have a docker container with ROS 2 `humble` in which the entire software runs (Realsense ros2 node+ open_vins). The D455 is damped to reduce the effect of vibrations during flight. 

Now, the estimator is working fine.

Here are my config files.

**estimator_config.yaml**
```bash
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 1 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 50 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 40 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: true
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 2.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] # initial gyroscope bias guess
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] # initial accelerometer bias guess

# ==================================================================
[... 37 lines cut ...]

# camera noises and chi-squared threshold multipliers
up_msckf_sigma_px: 1
up_msckf_chi2_multipler: 1
up_slam_sigma_px: 1
up_slam_chi2_multipler: 1
up_aruco_sigma_px: 1
up_aruco_chi2_multipler: 1

# masks for our images
use_mask: false

# imu and camera spacial-temporal
# imu config should also have the correct noise values
relative_config_imu: "kalibr_imu_chain.yaml"
relative_config_imucam: "kalibr_imucam_chain.yaml"
```

Note that I set `try_zupt: true`. The default value is `false`

I also used Kalibr to calibrate the camera and imu. I used [allan_ros2](https://github.com/CruxDevStuff/allan_ros2) to compute the statistics of the D455 imu.

Now there is a little issue. I can observe a drift of about 20-30 cm form the origin (marked on the ground)  after flying around and going back to origin. Is this normal? How to tune the `ov_msckf` such that I minimuze this drift?

Thank you.

Userpc1010, 2023-09-03:
What about using the calibration from the EEPROM of the camera itself. The depth camera was calibrated at the factory, is the intel calibration procedure more accurate? Also, if you are using a camera on a drone, it is probably better to select the stereo mode for greater tracking stability during fast movements, in which case calibration is not needed at all, because the images of the stereo camera are read already undistortions and the coefficients should be equal to 0.

mzahana, 2023-09-03:
@Userpc1010  I didn't use the camera params that are in the EEPROM, so I can't tell the difference. For the question related to the stability of stereo setup, I will leave it to @goldbattle  to answer.

Userpc1010, 2023-09-05:
@mzahana  I launched the camera with your settings, everything seems to be in order and the camera works a stable in stereo. Also I didn't have stable tracking without `try_zupt: true ` in mono or stereo maybe this should be added by default to rs_d455 @goldbattle ?
[image]

mzahana, 2023-09-12:
@Userpc1010 what was your configuration in mono mode that worked well for you?

Userpc1010, 2023-09-13:
To test mono I used the following configuration:

estimator_config.yaml:

```
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: false # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 1 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 50 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 40 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: true
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 1.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] # initial gyroscope bias guess
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] # initial accelerometer bias guess

# ==================================================================
# ==================================================================

record_timing_information: false # if we want to record timing information of the method
[... 269 lines cut ...]
    <!-- play the dataset -->
    <group if="$(arg dobag)">
        <node pkg="rosbag" type="play" name="rosbag" args="-d 1 -r $(arg bag_rate) -s $(arg bag_start) $(arg bag)" required="true"/>
    </group>

    <!-- record the trajectory if enabled -->
    <group if="$(arg dosave)">
        <node name="recorder_estimate" pkg="ov_eval" type="pose_to_file" output="screen" required="true">
            <param name="topic"      type="str" value="/ov_msckf/poseimu" />
            <param name="topic_type" type="str" value="PoseWithCovarianceStamped" />
            <param name="output"     type="str" value="$(arg path_est)" />
        </node>
    </group>

    <!-- path viz of aligned gt -->
    <group if="$(arg dolivetraj)">
        <node name="live_align_trajectory" pkg="ov_eval" type="live_align_trajectory" output="log" clear_params="true">
            <param name="alignment_type" type="str" value="posyaw" />
            <param name="path_gt"        type="str" value="$(arg path_gt)" />
        </node>
    </group>

</launch>
```
I conducted a test on a [car](https://github.com/rpng/open_vins/issues/379#issue-1885709490) stereo mode and got a result similar to some other frameworks ([1](https://github.com/introlab/rtabmap/issues/1061#issue-1764007522) [2](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/722#issuecomment-1573929822) [3](https://github.com/JuanTarrio/rebvo/issues/20#issuecomment-1693776906) [4](https://youtu.be/F8bkD8_AMDA)) that I tested before, but despite this, OpenVins is quite sophisticated and many problems have been solved here, such as resistance to rapid rotations and motion blur, imu init, which is necessary for drones.

Userpc1010, 2023-09-13:
@mzahana No, I didn't test mono on the machine, I did a mono test to make sure it was less stable and chose stereo. The stereo mode has better calibration and more accurately determines the image scale, I believe mono cannot surpass stereo.

goldbattle, 2023-09-14:
To both of you, could you post a bag and config file so I can take a look? @Userpc1010 could you open a new issue with the bag and configs for your driving dataset as I am interested in seeing what the problem there is. I don't think there should be a large problem filming from inside the car, but have you tried collecting a dataset with the sensor outside the vehicle?

Userpc1010, 2023-09-14:
@goldbattle  [Here](https://github.com/rpng/open_vins/issues/379#issue-1885709490) I published everything I have. Unfortunately I don't have a tripod to mount the camera outside the car.

goldbattle, 2023-10-28:
For now I don't want to enable the zvupt as default true since this can cause trouble for users just using the device handheld. Feel free to reopen if there is some problem with the current code.

closed 2023-10-28 by goldbattle