rpng/open_vins #374: Filter shutsdown in flight

mzahana, 2023-08-22:
Hi @goldbattle 

I am using the open_vins on a drone with D455+Jetson orin NX. It was working fine in flight until the ros node died with the following terminal output

```bash
Propagator::propagate_and_clone(std::shared_ptr<ov_msckf::State>, double): Assertion `std::abs((time1 - time0) - dt_summed) < 1e-4' failed.
[ERROR] [run_subscribe_msckf-1]: process has died [pid 46773, exit code -6, cmd '/home/admin/shared_volume/ros2_ws/install/ov_msckf/lib/ov_msckf/run_subscribe_msckf --ros-args -r __ns:=/ov_msckf --params-file /tmp/launch_params_aqezrtms --params-file /tmp/launch_params_mnuwltez --params-file /tmp/launch_params_0_irhk6i --params-file /tmp/launch_params_vcq38122 --params-file /tmp/launch_params_loc7x95u'].
[run_subscribe_msckf-1] [ZUPT]: passed disparity (0.126 <

```

estimator_config.yaml
```yaml
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 2 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 50 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 40 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: true
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 1.0 # how many seconds to collect initialization information
init_imu_thresh: 0.3 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 2.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] # initial gyroscope bias guess
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] # initial accelerometer bias guess

# ==================================================================
# ==================================================================

record_timing_information: false # if we want to record timing information of the method
record_timing_filepath: "/tmp/traj_timing.txt" # https://docs.openvins.com/eval-timing.html#eval-ov-timing-flame

# if we want to save the simulation state and its diagional covariance
# use this with rosrun ov_eval error_simulation
save_total_state: false
filepath_est: "/tmp/ov_estimate.txt"
filepath_std: "/tmp/ov_estimate_std.txt"
filepath_gt: "/tmp/ov_groundtruth.txt"

# ==================================================================
# ==================================================================

# our front-end feature tracking parameters
# we have a KLT and descriptor based (KLT is better implemented...)
use_klt: true # if true we will use KLT, otherwise use a ORB descriptor + robust matching
num_pts: 100 # number of points (per camera) we will extract and try to track
fast_threshold: 30 # threshold for fast extraction (warning: lower threshs can be expensive)
grid_x: 5 # extraction sub-grid count for horizontal direction (uniform tracking)
grid_y: 5 # extraction sub-grid count for vertical direction (uniform tracking)
min_px_dist: 15 # distance between features (features near each other provide less information)
knn_ratio: 0.70 # descriptor knn threshold for the top two descriptor matches
track_frequency: 31.0 # frequency we will perform feature tracking at (in frames per second / hertz)
downsample_cameras: false # will downsample image in half if true
num_opencv_threads: 4 # -1: auto, 0-1: serial, >1: number of threads
histogram_method: "HISTOGRAM" # NONE, HISTOGRAM, CLAHE

# aruco tag tracker for the system
# DICT_6X6_1000 from https://chev.me/arucogen/
use_aruco: false
num_aruco: 1024
downsize_aruco: true

# ==================================================================
# ==================================================================

# camera noises and chi-squared threshold multipliers
up_msckf_sigma_px: 1
up_msckf_chi2_multipler: 1
up_slam_sigma_px: 1
up_slam_chi2_multipler: 1
up_aruco_sigma_px: 1
up_aruco_chi2_multipler: 1

# masks for our images
use_mask: false

# imu and camera spacial-temporal
# imu config should also have the correct noise values
relative_config_imu: "kalibr_imu_chain.yaml"
relative_config_imucam: "kalibr_imucam_chain.yaml"
```

What could be the issue here?

goldbattle, 2023-08-23:
This should only happen if there are not enough IMU readings or the readings are dropped. I recommend inspecting this IMU stream to ensure you have a good connection to your sensor with sufficient bandwidth.

goldbattle, 2023-08-25:
> Thanks @goldbattle <https://github.com/goldbattle> for replying.
>
> I am using Realsense D455 camera.
>
> Here is the launch file I use to run the camera
>
> import launchfrom launch_ros.actions import ComposableNodeContainer, Nodefrom launch_ros.descriptions import ComposableNode
>
> def generate_launch_description():
>     """Launch file which brings up visual slam node configured for RealSense."""
>     realsense_camera_node = Node(
>         name='camera',
>         namespace='camera',
>         package='realsense2_camera',
>         executable='realsense2_camera_node',
>         parameters=[{
>                 'align_depth.enable': False,
>                 'enable_infra1': True,
>                 'enable_infra2': True,
>                 'enable_color': False,
>                 'enable_depth': False,
>                 'depth_module.emitter_enabled': 0,
>                 'rgb_camera.profile': '640x480x30',
>                 'depth_module.profile': '640x480x30',
>                 'enable_gyro': True,
>                 'enable_accel': True,
>                 'gyro_fps': 200,
>                 'accel_fps': 200,
>                 'unite_imu_method': 2
>         }]
>     )
>
>
>     return launch.LaunchDescription([realsense_camera_node])
>
> I even disabled all streams except the infra1/2 and imu.
>
> I also used ros2 topic hz --window 200 /camera/imu to check the frequency
> of the imu topic, and it does not go below 200hz for more than the time
> period the filter worked before it shut down.
>
> So, I am not sure if it is really a hardware problem.

Please record a ROS bag which is able to reproduce the issue and I can take
a look!

mzahana, 2023-09-03:
I was able to find the main issue. I had the the following setup. A Jetson Orin NX on a drone taht has `open_vins` installed inside a docker container with ros2 `humble`, and a laptop that has a docker container with ros2 `humble`. Both are connected to the same local network.  The problem (`ov_msckf` consistently crashing) happened when I run RViz2 on the laptop, and it starts to subscribe to the topics published on the Jetson Orin NX.

After some debugging, I realized that I was using two different DDS frameworks (cyclonedds on one device, and fastrtsp on the other). After fixing this issue, I am not getting this issue any more. Now, I am using fastrtsp on both. So, it was communication issue.
I am closing this as the issues is solved.

closed 2023-09-03 by mzahana