rpng/open_vins #375: Erase lost features from marg points, why writeing like this?

WZG3661, 2023-08-26:
```C++
// We also need to make sure that the max tracks does not contain any lost features
// This could happen if the feature was lost in the last frame, but has a measurement at the marg timestep
it1 = feats_lost.begin();
while (it1 != feats_lost.end()) {
  if (std::find(feats_marg.begin(), feats_marg.end(), (*it1)) != feats_marg.end()) {
    // PRINT_WARNING(YELLOW "FOUND FEATURE THAT WAS IN BOTH feats_lost and feats_marg!!!!!!\n" RESET);
    it1 = feats_lost.erase(it1);
  } else {
    it1++;
  }
}
```
If you want to erase lost feauters from marg points, why not writing like:
```C++
it1 = feats_marg.begin();
while (it1 != feats_marg.end()) {
  if (std::find(feats_lost.begin(), feats_lost.end(), (*it1)) != feats_lost.end()) {
    it1 = feats_marg.erase(it1);
  } else {
    it1++;
  }
}
```

WZG3661, 2023-08-26:
In my dataset, it makes a little different, but I can't which is better

goldbattle, 2023-09-14:
This shouldn't make any difference. We only insert SLAM features which have reached max tracks, so the feature shouldn't be used in either case. I do think the second you have posted would make clearer sense to someone reading through the code though..

closed 2023-09-22 by WZG3661