rpng/open_vins #376: Timeshift_cam_imu weird comportement

Atohme, 2023-08-29:
Hi,
I am working around openvins with a custom hardware around Ardupilot.

The behavior happen in the camera Imu configuration file.

My Kalibr calibration report me a time offset of -0.112xxxxx.
When I put those value in the timeshift_cam_Imu parameter the ov_msckf crashed. 

The error is removed by being under -0.1 of time offset,i.e, if I put -0.09 the ov_msckf doesn't crashed and the odometry is working perfectly. 

Also, enabling the time offset online optimization of openvins report me the same value of -0.11xxx. So most likely something is wrong in this parameter.

I hope the dev can help me with that.

goldbattle, 2023-09-06:
Please provide the configuration files along with a dataset which can
reproduce this.
Additionally, please post your calibration data and results from Kalibr
(pdf and txt along with the command you have run).

goldbattle, 2023-09-14:
Feel free to reopen if you have the dataset with the problem.

closed 2023-09-14 by goldbattle