rpng/open_vins #377: Tuning ov_msckf to minimize drift

mzahana, 2023-09-04:
Hi @goldbattle 

I am using `ov_msckf` on a drone, and it generally works OK. However, when I fly a course of around 10 meters with low speed, and then come back to the origin (marked on the ground), I observe a drift in the estimation. It is around 30cm. I am using the D455 camera with Jetson Orin NX (16 GB RAM) compute board so I believe I have enough compute power. How can I tune that `ov_msckf` parameters to minimize such drift.

For reference, here is the `estimator_config.yaml` I use.

```yaml
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 1 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 50 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 40 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: true
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 2.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] # initial gyroscope bias guess
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] # initial accelerometer bias guess

# ==================================================================
# ==================================================================

record_timing_information: false # if we want to record timing information of the method
record_timing_filepath: "/tmp/traj_timing.txt" # https://docs.openvins.com/eval-timing.html#eval-ov-timing-flame

# if we want to save the simulation state and its diagional covariance
# use this with rosrun ov_eval error_simulation
save_total_state: false
filepath_est: "/tmp/ov_estimate.txt"
filepath_std: "/tmp/ov_estimate_std.txt"
filepath_gt: "/tmp/ov_groundtruth.txt"

# ==================================================================
# ==================================================================

# our front-end feature tracking parameters
# we have a KLT and descriptor based (KLT is better implemented...)
use_klt: true # if true we will use KLT, otherwise use a ORB descriptor + robust matching
num_pts: 200 # number of points (per camera) we will extract and try to track
fast_threshold: 30 # threshold for fast extraction (warning: lower threshs can be expensive)
grid_x: 5 # extraction sub-grid count for horizontal direction (uniform tracking)
grid_y: 5 # extraction sub-grid count for vertical direction (uniform tracking)
min_px_dist: 15 # distance between features (features near each other provide less information)
knn_ratio: 0.70 # descriptor knn threshold for the top two descriptor matches
track_frequency: 31.0 # frequency we will perform feature tracking at (in frames per second / hertz)
downsample_cameras: false # will downsample image in half if true
num_opencv_threads: 4 # -1: auto, 0-1: serial, >1: number of threads
histogram_method: "HISTOGRAM" # NONE, HISTOGRAM, CLAHE

# aruco tag tracker for the system
# DICT_6X6_1000 from https://chev.me/arucogen/
use_aruco: false
num_aruco: 1024
downsize_aruco: true

# ==================================================================
# ==================================================================

# camera noises and chi-squared threshold multipliers
up_msckf_sigma_px: 1
up_msckf_chi2_multipler: 1
up_slam_sigma_px: 1
up_slam_chi2_multipler: 1
up_aruco_sigma_px: 1
up_aruco_chi2_multipler: 1

# masks for our images
use_mask: false

# imu and camera spacial-temporal
# imu config should also have the correct noise values
relative_config_imu: "kalibr_imu_chain.yaml"
relative_config_imucam: "kalibr_imucam_chain.yaml"
```

Thank you.

goldbattle, 2023-09-06:
> # aruco tag tracker for the system# DICT_6X6_1000 from https://chev.me/arucogen/use_aruco: falsenum_aruco: 1024downsize_aruco: true

I don't see any big issues with what you posted.
Here are some thoughts.

If the robotic platform is doing constrained motion (1-axis rotation), then
likely you should fix the extrinsic calibration.
You can first try to do a general trajectory by hand to see if this gives
you a good refined calibration (or just use an offline optimized one from
Kalibr).
Then fix the extrinsics.

Intrinsics and timeoffset should be able to be calibrated ok.

How did you get your IMU noise parameters?
Likely if you are on a MAV, then motor noise is going to affect these.
Did you calibrate the IMU intrinsic parameters and provide these?

mzahana, 2023-09-08:
@goldbattle Thanks for your reply.
I used Kalibr for the offline calibration.

Here are the results.
`d455_stereo_imu-camchain-imucam.yaml`
```yaml
cam0:
  T_cam_imu:
  - [0.9999958955314554, -0.0026986387404283703, 0.0009624287980268205, 0.031988892285795864]
  - [0.0026994985594231804, 0.9999959574359712, -0.0008932072788633144, -0.008702065374357535]
  - [-0.0009600144635808081, 0.0008958016878759553, 0.9999991379554113, -0.02015716670564154]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [1]
  camera_model: pinhole
  distortion_coeffs: [0.006810709850531221, -0.00035774484224898915, 0.000163275469486564, -0.0013628846487450934]
  distortion_model: radtan
  intrinsics: [392.276461026783, 392.2076415717271, 320.8482141777144, 241.01828007669283]
  resolution: [640, 480]
  rostopic: /camera/infra1/image_rect_raw
  timeshift_cam_imu: 0.005199107334216512
cam1:
  T_cam_imu:
  - [0.9999740517275426, -0.002762307229153594, 0.006653234579798995, -0.06199981834475254]
  - [0.0027743189584098276, 0.9999945372365269, -0.0017968448518782768, -0.008748107900169019]
  - [-0.006648234797228143, 0.0018152564216879809, 0.9999762526271335, -0.019826904661799198]
  - [0.0, 0.0, 0.0, 1.0]
  T_cn_cnm1:
  - [0.9999838050993602, -6.881026780535209e-05, 0.005690764812071861, -0.09387408167006568]
  - [7.395423767066402e-05, 0.999999588919511, -0.0009037098949213374, -6.662804820914188e-05]
  - [-0.005690700288189607, 0.0009041161156028242, 0.9999833991143423, 0.0005198342931163642]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [0]
  camera_model: pinhole
  distortion_coeffs: [0.006089577932177055, 0.0006761464415174285, 0.00034541579726369176, -0.002411714893977257]
  distortion_model: radtan
  intrinsics: [391.94563266141654, 392.194156418658, 318.6227610379495, 241.19594646363703]
  resolution: [640, 480]
  rostopic: /camera/infra2/image_rect_raw
  timeshift_cam_imu: 0.005200396803312446
```
`d455_stereo_imu-imu.yaml`
```yaml
imu0:
  T_i_b:
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
  accelerometer_noise_density: 0.018325460766195713
  accelerometer_random_walk: 0.00315265486799695
  accelerometers:
    M:
    - [0.9850005161541593, 0.0, 0.0]
    - [-0.0010659744348495597, 0.9928231590544696, 0.0]
    - [0.0029035502259805745, 0.022543809972056676, 1.0016290323214185]
  gyroscope_noise_density: 0.005658445654838128
  gyroscope_random_walk: 0.0004924482796880002
  gyroscopes:
    A:
    - [1.4028016822296411e-05, -0.0004093748694620693, 6.83514932677649e-05]
    - [0.0001130653242100852, 0.0016182902003747013, -0.0003721349646165901]
    - [0.00014459152269515854, -8.962959661707025e-05, 6.527559317237045e-05]
    C_gyro_i:
    - [0.9999899802473953, -0.004167092535127969, -0.0016354646489026166]
    - [0.004167688113960002, 0.9999912500158429, 0.00036092594374198517]
    - [0.001633946326806948, -0.00036773843393135346, 0.9999985974929393]
    M:
    - [1.0252682843452035, 0.0, 0.0]
    - [-0.009317074064285835, 1.0230957936554894, 0.0]
    - [-0.0070358516203933375, 0.00319988649549574, 0.9995381578198]
  model: scale-misalignment
  rostopic: /camera/imu
  time_offset: 0.0
  update_rate: 200.0
```

`imu.yaml` I used allan_ros2 to compute the IMU statistics.
```yaml
# D455 imu data
#Accelerometer
#accelerometer_noise_density: 0.0036650921532391427 
#accelerometer_random_walk: 0.00031526548679969497 

#Gyroscope
#gyroscope_noise_density: 0.0011316891309676256 
#gyroscope_random_walk: 4.9244827968800026e-05 

# density x5
# walk x10
#Accelerometer
accelerometer_noise_density: 0.018325460766195713 
accelerometer_random_walk: 0.0031526548679969497 
#Gyroscope
gyroscope_noise_density: 0.005658445654838128
gyroscope_random_walk: 4.9244827968800026e-04 

rostopic: '/camera/imu' #Make sure this is correct
update_rate: 200.0 #Make sure this is correct

```

mzahana, 2023-09-08:
@goldbattle  I also attach the PDF report generated by Kalibr after the calibration.
[d455_stereo_imu-report-imucam.pdf](https://github.com/rpng/open_vins/files/12554336/d455_stereo_imu-report-imucam.pdf)

goldbattle, 2023-09-08:
How do your IMU noises compare to the example in the config folder for the
d455?
Have you tried increasing a bit more (maybe x10?) to handle the extra
vibrations?

If you are able to post a bag that could help me debug locally.

goldbattle, 2023-09-08:
That is a very nice looking calibration. Impressed you got such good
reprojection errors!

javieryu, 2023-10-12:
@mzahana Could you link to or post your configuration, and any tips that you have? I have the exact same setup (16Gb Orin + d455), and I am struggling to get any working result. The d455 mono configuration that was included with open_vins also did not work for me.

mzahana, 2023-10-14:
@javieryu Most likely you need to have a good camera/IMU calibration. I suggest that you use Kalibr for this task.

javieryu, 2023-10-17:
@Userpc1010 @mzahana I'll try those ideas. Thanks!

goldbattle, 2023-11-22:
Feel free to reopen with a bag and config YAML files to launch it with. Otherwise I am not sure how to debug any more.

closed 2023-11-22 by goldbattle