rpng/open_vins #379: Drift trajectory in open space?

Userpc1010, 2023-09-07:
What is the reason for the occurrence of drift in open spaces in various navigation frameworks? I tested 5 different frameworks realtime on the car in different modes (vins-rgb-d mono+rgb-d mode, orb-slam3 rgb-d-inertial, rebivo mono, rtab-map rgb-d, openvins stereo) and everywhere I get a similar result and strong drift effects?

Unfortunately, I could not set up the r-viz visualizer well, the trajectory on the record can be very poorly visible((

https://youtu.be/LlvSzjQ40l4

[Rosbag_all_topic_d455](https://drive.google.com/file/d/1SMOyNVFLOwdwerfKEA-wXF7bNstNCnXx/view?usp=drive_link)

[Short_tf_static_rqt](https://drive.google.com/file/d/1U_BiApcbnSDN7oTeuZJgKcJaQ9E91zmT/view?usp=sharing)

goldbattle, 2023-09-14:
I am not sure. Do you have the configuration files you used to launch OpenVINS? Does the feature tracking look good? Does this sensor work for general 3D motion for you with good quality?

If you are running on a car dataset, I recommend using a mask to remove features from the dash. See the rpng_ironsides dataset for and example car dataset.
- https://github.com/rpng/open_vins/tree/master/config/rpng_ironsides
- https://docs.openvins.com/gs-datasets.html#gs-data-rpng

Userpc1010, 2023-09-14:
My configuration was like this:

estimator_config.yaml:

```
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 2 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 50 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 40 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: true
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 1.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0, 0.0, 0.0 ] # initial gyroscope bias guess
init_dyn_bias_a: [ 0.0, 0.0, 0.0 ] # initial accelerometer bias guess

# ==================================================================
# ==================================================================

record_timing_information: false # if we want to record timing information of the method
[... 273 lines cut ...]

    <!-- record the trajectory if enabled -->
    <group if="$(arg dosave)">
        <node name="recorder_estimate" pkg="ov_eval" type="pose_to_file" output="screen" required="true">
            <param name="topic"      type="str" value="/ov_msckf/poseimu" />
            <param name="topic_type" type="str" value="PoseWithCovarianceStamped" />
            <param name="output"     type="str" value="$(arg path_est)" />
        </node>
    </group>

    <!-- path viz of aligned gt -->
    <group if="$(arg dolivetraj)">
        <node name="live_align_trajectory" pkg="ov_eval" type="live_align_trajectory" output="log" clear_params="true">
            <param name="alignment_type" type="str" value="posyaw" />
            <param name="path_gt"        type="str" value="$(arg path_gt)" />
        </node>
    </group>

</launch>
```
Unfortunately, I did not pay attention to the tracking functions, I mainly focused on the trajectory of the visualizer, but I think this can be reproduced using the ROS bag attached to the message, if your don’t mind checking it out, I’m just starting to understand openvins.

This is a good question, I'm probably not so savvy as to give a definite answer, I used an intel realsense d455 in stereo mode.

Can I make a homemade mask to cover the bonnet? I suppose this will not solve all the problems, for example, the re-reflection of light from the bonnet onto the camera may disrupt the exposure, but unfortunately I do not have a tripod for external mounting of the camera(((.

Userpc1010, 2024-02-23:
I did a small test where camera mount in rover (without the front hood on a car), but the results were worse than previous attempts.

launch_d455.example
```

<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <arg name="serial_no"           default="215122253909"/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d455"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="respawn"             default="true"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="false"/>

  <arg name="infra_width"        default="848"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="200"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="filters"                   default="pointcloud"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="unite_imu_method"          default="copy"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>
      <arg name="respawn"                  value="$(arg respawn)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
[... 259 lines cut ...]
  camera_model: pinhole
  distortion_coeffs: [0.0, 0.0, 0.0, 0.0]
  distortion_model: radtan
  intrinsics: [431.411, 431.411, 429.483, 240.654]
  resolution: [848, 480]
  rostopic: /d455/infra2/image_rect_raw
  timeshift_cam_imu: 0.005012709485041032
  
#cam0:
#T_cam_imu:
#   - [0.9999654398038452, 0.007342326779113337, -0.003899927610975742, -0.027534314618518095]
#   - [-0.0073452195116216765, 0.9999727585590525, -0.0007279355223411334, -0.0030587146933711722]
#   - [0.0038944766308488753, 0.0007565561891287445, 0.9999921303062861, -0.023605118842939803]
#   - [0.0, 0.0, 0.0, 1.0]
# cam_overlaps: []
# camera_model: pinhole
# distortion_coeffs: [-0.045761895748285604, 0.03423951132164367, -0.00040139057556727315, 0.000431371425853453]
# distortion_model: radtan
# intrinsics: [416.85223429743274, 414.92069080087543, 421.02459311003213, 237.76180565241077]
# resolution: [848, 480]
# rostopic: /d455/color/image_raw
# timeshift_cam_imu: 0.002524377913673846
```

DIY Rover Test OpenVINS https://youtu.be/lIQRj7pLs2k

closed 2024-02-23 by Userpc1010