rpng/open_vins #384: build failed on WSL Ubuntu 20.04 ROS 1 neotic

kruzel, 2023-09-20:
CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterMSCKF.cpp.o] Error 1

[build.make.000.log](https://github.com/rpng/open_vins/files/12670897/build.make.000.log)

[image]

goldbattle, 2023-09-21:
This is likely an out of memory error.
Not much I can do besides recommending adding a swapfile or building with
catkin build -j1
https://gist.github.com/susanBuck/b9310f8f0886105008a2f6a18a18d7ae

Shouldn't be an issue with WSL, but I have never tried this...

kruzel, 2023-09-25:
my VM has 32GB RAM,
should I increase?
Ofer Kruzel
M: +972 (54) 4516627

kruzel, 2023-09-26:
works on non VM matchine

closed 2023-09-26 by kruzel