rpng/open_vins #385: Colcon build failed ov_init

Genozen, 2023-09-26:
Prior to this, I had to build Ceres from source, but later I was able to use `sudo apt` to reinstall Ceres.

My workspace dir includes:
- build
- ceres-solver
- install
- log
- src

```
colcon build --parallel-workers 1
Starting >>> ov_core 
Finished <<< ov_core [0.18s]                     
Starting >>> ov_init
--- stderr: ov_init                               
In file included from /home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.cpp:22:
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:32:64: error: expected class-name before ‘{’ token
   32 | class State_JPLQuatLocal : public ceres::LocalParameterization {
      |                                                                ^
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:44:8: error: ‘bool ov_init::State_JPLQuatLocal::Plus(const double*, const double*, double*) const’ marked ‘override’, but does not override
   44 |   bool Plus(const double *x, const double *delta, double *x_plus_delta) const override;
      |        ^~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:57:8: error: ‘bool ov_init::State_JPLQuatLocal::ComputeJacobian(const double*, double*) const’ marked ‘override’, but does not override
   57 |   bool ComputeJacobian(const double *x, double *jacobian) const override;
      |        ^~~~~~~~~~~~~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:59:7: error: ‘int ov_init::State_JPLQuatLocal::GlobalSize() const’ marked ‘override’, but does not override
   59 |   int GlobalSize() const override { return 4; };
      |       ^~~~~~~~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:61:7: error: ‘int ov_init::State_JPLQuatLocal::LocalSize() const’ marked ‘override’, but does not override
   61 |   int LocalSize() const override { return 3; };
      |       ^~~~~~~~~
gmake[2]: *** [CMakeFiles/ov_init_lib.dir/build.make:132: CMakeFiles/ov_init_lib.dir/src/ceres/State_JPLQuatLocal.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/dynamic/DynamicInitializer.cpp:27:
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:32:64: error: expected class-name before ‘{’ token
   32 | class State_JPLQuatLocal : public ceres::LocalParameterization {
      |                                                                ^
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:44:8: error: ‘bool ov_init::State_JPLQuatLocal::Plus(const double*, const double*, double*) const’ marked ‘override’, but does not override
   44 |   bool Plus(const double *x, const double *delta, double *x_plus_delta) const override;
      |        ^~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:57:8: error: ‘bool ov_init::State_JPLQuatLocal::ComputeJacobian(const double*, double*) const’ marked ‘override’, but does not override
   57 |   bool ComputeJacobian(const double *x, double *jacobian) const override;
      |        ^~~~~~~~~~~~~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:59:7: error: ‘int ov_init::State_JPLQuatLocal::GlobalSize() const’ marked ‘override’, but does not override
   59 |   int GlobalSize() const override { return 4; };
      |       ^~~~~~~~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/State_JPLQuatLocal.h:61:7: error: ‘int ov_init::State_JPLQuatLocal::LocalSize() const’ marked ‘override’, but does not override
   61 |   int LocalSize() const override { return 3; };
      |       ^~~~~~~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/dynamic/DynamicInitializer.cpp: In member function ‘bool ov_init::DynamicInitializer::initialize(double&, Eigen::MatrixXd&, std::vector<std::shared_ptr<ov_type::Type> >&, std::shared_ptr<ov_type::IMU>&, std::map<double, std::shared_ptr<ov_type::PoseJPL> >&, std::unordered_map<long unsigned int, std::shared_ptr<ov_type::Landmark> >&)’:
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/dynamic/DynamicInitializer.cpp:673:30: error: no matching function for call to ‘ceres::Problem::AddParameterBlock(double*&, int, ov_init::State_JPLQuatLocal*&)’
  673 |     problem.AddParameterBlock(var_ori, 4, ceres_jplquat);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/ceres/ceres.h:64,
                 from /home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/Factor_GenericPrior.h:25,
                 from /home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/dynamic/DynamicInitializer.cpp:24:
/usr/local/include/ceres/problem.h:259:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int)’
  259 |   void AddParameterBlock(double* values, int size);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:259:8: note:   candidate expects 2 arguments, 3 provided
/usr/local/include/ceres/problem.h:271:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int, ceres::Manifold*)’
  271 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:271:62: note:   no known conversion for argument 3 from ‘ov_init::State_JPLQuatLocal*’ to ‘ceres::Manifold*’
  271 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |                                                    ~~~~~~~~~~^~~~~~~~
/home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/dynamic/DynamicInitializer.cpp:766:32: error: no matching function for call to ‘ceres::Problem::AddParameterBlock(double*&, int, ov_init::State_JPLQuatLocal*&)’
  766 |       problem.AddParameterBlock(var_calib_ori, 4, ceres_calib_jplquat);
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/ceres/ceres.h:64,
                 from /home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/ceres/Factor_GenericPrior.h:25,
                 from /home/genozen/Desktop/ros2_openvins/src/open_vins/ov_init/src/dynamic/DynamicInitializer.cpp:24:
/usr/local/include/ceres/problem.h:259:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int)’
  259 |   void AddParameterBlock(double* values, int size);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:259:8: note:   candidate expects 2 arguments, 3 provided
/usr/local/include/ceres/problem.h:271:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int, ceres::Manifold*)’
  271 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:271:62: note:   no known conversion for argument 3 from ‘ov_init::State_JPLQuatLocal*’ to ‘ceres::Manifold*’
  271 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |                                                    ~~~~~~~~~~^~~~~~~~
gmake[2]: *** [CMakeFiles/ov_init_lib.dir/build.make:160: CMakeFiles/ov_init_lib.dir/src/dynamic/DynamicInitializer.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:143: CMakeFiles/ov_init_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ov_init [17.5s, exited with code 2]

Summary: 1 package finished [17.8s]
  1 package failed: ov_init
  1 package had stderr output: ov_init
  4 packages not processed
```

goldbattle, 2023-09-26:
Looks like a bad version of ceres. What version are you using? Did you
install libceres-dev?

goldbattle, 2023-10-28:
This is due to a backwards incompatible change of ceres-solver.
> LocalParameterization has been removed, use [Manifold](http://ceres-solver.org/nnls_modeling.html#_CPPv4N5ceres8ManifoldE) instead.
http://ceres-solver.org/version_history.html#id1

You should install a version earlier than 2.2.0

goldbattle, 2024-02-12:
I will keep this open till I am able to address it. I think maybe a compatibility layer or something is needed to make this work with both versions.

bryangd34, 2024-06-07:
@goldbattle 
Are there any plans to update OpenVINS so that is is compatible with Ceres 2.2.0?

I am having issues installing older Ceres versions from source on Ubuntu 24.

gregoryw3, 2025-01-15:
Hello!

Originally I wanted to try updating open_vins to support Ceres 2.1.0 as the release notes stated that LocalParameterization was removed and Manifold was a drop in replacement. I think that's the case from a quick glance but I have never used open_vins or ceres so not sure what changes would be appropriate. 

I was able to compile Ceres 2.0.0 (clang) on Ubuntu arm64 24.04 (M1 Macbook running VM under Parallels by doing the following:

First follow the install steps at https://docs.openvins.com/gs-installing.html

In: ceres-solver/cmake/FindTBB.cmake
```cmake
 #only read the start of the file
 file(STRINGS
      "${TBB_INCLUDE_DIR}/tbb/version.h"
      TBB_VERSION_CONTENTS
      REGEX "VERSION")
```
Thanks to https://github.com/ceres-solver/ceres-solver/issues/1036 and https://github.com/yppah56/OpenFPGA/commit/8cdaceb4358c3590e473bf7d623c46ab932e2e6a

In: ceres-solver/internal/ceres/covariance_impl.cc
```cpp
  const SuiteSparse_long rank = SuiteSparseQR<double, SuiteSparse_long>(SPQR_ORDERING_BESTAMD,
                                                      SPQR_DEFAULT_TOL,
                                                      cholmod_jacobian.ncol,
                                                      &cholmod_jacobian,
                                                      &R,
                                                      &permutation,
                                                      &cc);
```
I compiled it using `cmake --build .`
During the build process I only received these warnings:

```shell
[ 60%] Building CXX object internal/ceres/CMakeFiles/small_blas_gemv_benchmark.dir/small_blas_gemv_benchmark.cc.o
ceres-solver/internal/ceres/small_blas_gemv_benchmark.cc:81:6: warning: no previous declaration for ‘void ceres::BM_MatrixVectorMultiply(benchmark::State&)’ [-Wmissing-declarations]
   81 | void BM_MatrixVectorMultiply(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~
ceres-solver/internal/ceres/small_blas_gemv_benchmark.cc:97:6: warning: no previous declaration for ‘void ceres::BM_MatrixTransposeVectorMultiply(benchmark::State&)’ [-Wmissing-declarations]
   97 | void BM_MatrixTransposeVectorMultiply(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 60%] Linking CXX executable ../../bin/small_blas_gemv_benchmark
[ 60%] Built target small_blas_gemv_benchmark
[ 61%] Building CXX object internal/ceres/CMakeFiles/small_blas_gemm_benchmark.dir/small_blas_gemm_benchmark.cc.o
ceres-solver/internal/ceres/small_blas_gemm_benchmark.cc:89:6: warning: no previous declaration for ‘void ceres::internal::BM_MatrixMatrixMultiplyDynamic(benchmark::State&)’ [-Wmissing-declarations]
   89 | void BM_MatrixMatrixMultiplyDynamic(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ceres-solver/internal/ceres/small_blas_gemm_benchmark.cc:135:6: warning: no previous declaration for ‘void ceres::internal::BM_MatrixTransposeMatrixMultiplyDynamic(benchmark::State&)’ [-Wmissing-declarations]
  135 | void BM_MatrixTransposeMatrixMultiplyDynamic(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 61%] Linking CXX executable ../../bin/small_blas_gemm_benchmark
[ 61%] Built target small_blas_gemm_benchmark
[ 61%] Building CXX object internal/ceres/CMakeFiles/invert_psd_matrix_benchmark.dir/invert_psd_matrix_benchmark.cc.o
ceres-solver/internal/ceres/invert_psd_matrix_benchmark.cc:65:6: warning: no previous declaration for ‘void ceres::internal::BenchmarkDynamicallyInvertPSDMatrix(benchmark::State&)’ [-Wmissing-declarations]
   65 | void BenchmarkDynamicallyInvertPSDMatrix(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 62%] Linking CXX executable ../../bin/invert_psd_matrix_benchmark
[ 62%] Built target invert_psd_matrix_benchmark
[ 62%] Building CXX object internal/ceres/CMakeFiles/schur_eliminator_benchmark.dir/schur_eliminator_benchmark.cc.o
ceres-solver/internal/ceres/schur_eliminator_benchmark.cc:130:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorEliminate(benchmark::State&)’ [-Wmissing-declarations]
  130 | void BM_SchurEliminatorEliminate(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
ceres-solver/internal/ceres/schur_eliminator_benchmark.cc:153:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorBackSubstitute(benchmark::State&)’ [-Wmissing-declarations]
  153 | void BM_SchurEliminatorBackSubstitute(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ceres-solver/internal/ceres/schur_eliminator_benchmark.cc:181:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorForOneFBlockEliminate(benchmark::State&)’ [-Wmissing-declarations]
  181 | void BM_SchurEliminatorForOneFBlockEliminate(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ceres-solver/internal/ceres/schur_eliminator_benchmark.cc:195:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorForOneFBlockBackSubstitute(benchmark::State&)’ [-Wmissing-declarations]
  195 | void BM_SchurEliminatorForOneFBlockBackSubstitute(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

After running `sudo apt remove libceres-dev` , (which removed nav2 due to dependencies), I was able to compile the library (well with some apt install shenanigans with libceres-dev because of unrelated reasons, so build might not be reproducible).

I had to edit the following files:

In: open_vins/ov_msckf/src/ros/ROS2Visualizer.h
```cpp
#include <image_transport/image_transport.hpp>
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#include <cv_bridge/cv_bridge.hpp>
```

In: open_vins/ov_msckf/src/ros/ROS2VisualizerHelper.h
```cpp
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
```

Unfortunately I am not at a point where I can evaluate if this works correctly, however happy to respond to any questions!