rpng/open_vins #387: Test webcam example

Userpc1010, 2023-10-05:
How can I bypass the limitation and use only OpenCV: 4.7.0?

[image]

goldbattle, 2023-10-05:
Maybe you have not specified the correct camera. Do you have the camera ID
set correctly?
Are you able to compile the OpenCV explain, or run the other binaries from
OpenVINS?

Userpc1010, 2023-10-07:
Yes, apparently the problem was in the camera, at the moment I building openvins with librealsense directly (without ROS) and do not observe this problem.

[image]

closed 2023-10-13 by Userpc1010