rpng/open_vins #388: D455 (plug and play) demo

mcamurri, 2023-10-06:
Dear all,
nedless to say, OpenVINS is awesome.

I decided to try it quickly on my D455 unit on Ubuntu 20.04 and Focal.

I compiled it no problem by using all dependencies from apt.
Librealsense is correctly installed from apt following official Intel's instructions, then I have compiled `realsense-ros` with `catkin`.

Output of `rs-enumerate`:
```
Device info: 
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	044422250186
    Firmware Version              : 	5.15.1
    Recommended Firmware Version  : 	5.15.1
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video2
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	032623060726
    Firmware Update Id            : 	032623060726
```

Then and did the following:

1. plugged in the device
2. copied the `launch_d455.example` into `launch_d455.launch`, changing the serial number to `044422250186`
3. `roslaunch launch_d455.launch`
4. `roslaunch ov_msckf subscribe.launch config:=rs_d455`

I'd assumed the pre-defined config would be just fine for a quick demo but apparently there is some problems parsing the arguments:

```
process[ov_msckf-1]: started with pid [23473]
overriding node verbosity with value from ROS!
Setting printing level to: INFO
overriding node max_cameras with value from ROS!
overriding node max_cameras with value from ROS!
overriding node use_stereo with value from ROS!
parameter T_imu_cam not found, trying T_cam_imu instead (will return T_imu_cam still)!
the sensor cam1 of type [NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE] was not found...
the sensor cam1 of type [St6vectorIdSaIdEE] was not found...
the sensor cam1 of type [St6vectorIdSaIdEE] was not found...
the sensor cam1 of type [St6vectorIiSaIiEE] was not found...
the sensor cam1 of type [N5Eigen6MatrixIdLi4ELi4ELi0ELi4ELi4EEE] was not found...
overriding node record_timing_information with value from ROS!
overriding node record_timing_filepath with value from ROS!
overriding node use_stereo with value from ROS!
parameter T_imu_cam not found, trying T_cam_imu instead (will return T_imu_cam still)!
the sensor cam1 of type [NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE] was not found...
the sensor cam1 of type [St6vectorIdSaIdEE] was not found...
the sensor cam1 of type [St6vectorIdSaIdEE] was not found...
the sensor cam1 of type [St6vectorIiSaIiEE] was not found...
the sensor cam1 of type [N5Eigen6MatrixIdLi4ELi4ELi0ELi4ELi4EEE] was not found...
subscribing to IMU: /d455/imu
the sensor cam1 of type [NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE] was not found...
subscribing to cam (stereo): /d455/color/image_raw
subscribing to cam (stereo): /cam1/image_raw
unable to parse all parameters, please fix
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
================================================================================REQUIRED process [ov_msckf-1] has died!
process has died [pid 23473, exit code -6, cmd /home/mcamurri/catkin_ws/devel/lib/ov_msckf/run_subscribe_msckf __name:=ov_msckf __log:=/home/mcamurri/.ros/log/3a7b1664-6449-11ee-aa3d-33dbd841adcc/ov_msckf-1.log].
log file: /home/mcamurri/.ros/log/3a7b1664-6449-11ee-aa3d-33dbd841adcc/ov_msckf-1*.log
Initiating shutdown!
================================================================================
[ov_msckf-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```

any pointers to where to start debugging from?

Thanks!

goldbattle, 2023-10-06:
Also specify a single camera.

goldbattle, 2023-10-07:
Specifically this should work:
```
roslaunch ov_msckf subscribe.launch config:=rs_d455 max_cameras:=1
```

Or you will need to change the driver of the D455 and do calibration if you wish to leverage the stereo cameras (I believe it has stereo for the IR?). The config we have is just for the main color global shutter camera. Hope this helps.

mcamurri, 2023-10-11:
yep, that worked like a charm, thank you! 

Yes, the d455 has a stereo pair, and because I've always used the d435i which has a rolling shutter RGB camera, I thought the demo launchfile was configured for the stereo pair. 

If you find it useful, I can update the tutorials in a PR, but this issue can be closed.

closed 2023-10-29 by goldbattle