rpng/open_vins #390: Static initialization failed

huang-shijie, 2023-10-08:
When running the source code, it prompts that static initialization is always unsuccessful, and the terminal displays as follows:
[image]
[image]
how to solve it.

Aishkrish18, 2023-10-09:
In order to do static initialisation, the camera must be activated in all directions and orientations by giving enough initial movements. Maybe you are not activating all orientations enough?

huang-shijie, 2023-10-09:
thanks for you answer,i have solved this issue.
[image]

closed 2023-10-28 by goldbattle

goldbattle, 2024-03-25:
@Alex-Beh you should just need to pick it up from static.