rpng/open_vins #392: Some questions about state->_imu and state->_clones_IMU

gtx818, 2023-10-16:
Hello all,
I have some questions about the code. It seems that the state->_imu is only propagated in the EKFPropagation process. In the EKFUpdate process, only state->_clones_IMU are used for update, which means the state->_imu is not updated. However, I see that in the visualization, state->_IMU is utilized for displaying the estimated trajectory. Is there something that I ignored?

goldbattle, 2023-10-28:
You are correct in your observation that we don't use the `state->_imu`. There are actually two reasons. The first is that this simplifies our code to just a map of stochastic clones. Additionally the clones have been cloned at the "camera time" and have a non-identity Jacobian during cloning. So if we estimate the offset between the cameras we should not use the `_imu` as it is in the IMU time.

Another point, is that this cloning results in the `_imu` and the latest clone to be correlated. This means that if something updates the clone, then the `_imu` will be updated though the correlations. You can print out the `dx` yourself to inspect if both the latest clone and `_imu` are updated the exact same amount (if no camera-imu time offset is estimated).

So, there is no problem and everything is getting updated since the covariance is leveraged (see the Kalman gain update equations).

gtx818, 2023-11-01:
I understand, thank you! Very exquisite design!

closed 2023-11-22 by goldbattle