rpng/open_vins #394: Questions with OpenVINS Algorithm in Different Environments

Aishkrish18, 2023-10-17:
Apart from the issue mentioned in the #386 , I have encountered a few other challenges in the algorithm.

 1. I've been testing this algorithm in two distinct environments, an office space and a large shopfloor hall with a high ceiling(30-40 mtrs above). Does the testing environment matter or are there certain parameters that need to be tuned for this? On testing it in the huge hall, imu updates relatively slow since it captures features farther away and drifts a lot compare to tests conducted in the office space. How do i tackle this? 
 
2. In the estimator files - kalibr_imu_chain.yaml - what is T_i_b: how does it affect the estimation results?  

3. How does the camera resolution affect the estimation results, what is the optimal value that you would suggest? 

4. Although the camera streams images at 30 hz, the trackhist published by ov_msckf is always fixed to 20 hz, is it hardcoded, how can i change this? does this affect the update/estimation of the imu values, since imu always updates after the robot and is offseted behind as i move it from the start. 

Thanks in advance!

zakih, 2023-10-25:
Also interested in the camera/image resolution question (3rd question above)

Seeing bad performance with 1920x1080 images, but cutting that in half produces way better results

goldbattle, 2023-10-29:
You will get a performance difference based on the average depth of the scene. You might want to try to increase your sliding window to try to get more disparity to help with these larger environments. Additionally, using anchored inverse depth features can help here too.

`T_i_b` isn't used anywhere, and I believe is a historical relic of other systems that leveraged the output from Kalibr.

The majority of the algorithm is tuned for 640p. I know that much of the feature tracking frontend (which leverages OpenCV) suffers quite a lot at higher resolutions. You mileage may vary, and this is why there is a `downsample_cameras` which will cut the resolution in half if you have a high resolution sensor that you can't change.

You can increase the tracking rate in the config file. `track_frequency` should be what you are wanting.

I believe ov_secondary will only work with live sensor data. I know it has some compile issues that you might need to solve for newer OS versions (I last tested it on 18.04).

goldbattle, 2023-11-22:
Feel free to reopen if you have any more followup questions to my response!

closed 2023-11-22 by goldbattle