rpng/open_vins #396: Why is there such a strange quaternion in OpenVins?

Userpc1010, 2023-10-31:
In most applications that I usually see when the device is in its default angular position it returns the quaternion 1,0,0,0 however OpenVins sends me something different:

[image]

I want to visualize points from a state vector on a voxels map (white color cube), but due to the quaterninon shift needed to return it to its usual 1,0,0,0 appearance, these points do not completely coincide with the voxels map:

[image]

in rviz the points look something like this: 

[image]

Perhaps I should change the mathematics of vectors rotation using a quaternion to use a quaternion from OpenVins, but due to the non-commutative nature of multiplication, it is not obvious how to do this?

goldbattle, 2023-10-31:
We use a JPL quaternion representation.
http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf

We estimate the R_GtoI / q_GtoI rotation.

So if you take the xyzw from OpenVINS and use it in a python or other
library (e.g. Eigen3), then you will then get the q_ItoG rotation.
There are some comments in the ROS visualizer class about this transform if
you are interested.
https://github.com/rpng/open_vins/blob/master/ov_msckf/src/ros/ROS1Visualizer.cpp#L329-L339

Things published onto ROS should be correct.

Hope this helps.

Userpc1010, 2023-11-01:
I used Eigen and QT functionality to do a quaternion to rotation matrix conversion and got a similar result - the matrix shows some rotation while the device is in the default position:

```
QQuaternion(scalar:0.477802, vector:(-0.518875, 0.504593, -0.497856))
QGenericMatrix<3, 3, float>(
-0.00494754-0.0478885   0.99884         
 -0.999394-0.0341834-0.00658908         
 0.0344592 -0.998268-0.0476904         
)
Eigen quat to matrix 3x3 result:
   -0.00494754    -0.0478885    0.99884
   -0.999394    -0.0341834    -0.00658908
   0.0344592    -0.998268    -0.0476904
```
While the usual representation looks like this using Machony or Magwick filter from imu:

 ```
 0.999656    0.0261564    0.00130503
   -0.0261802    0.99937    0.0238789
   -0.000677511    -0.0239049    0.999713
```
Those, when the device starts, its angular position is a unit matrix, which is familiar and understandable to me.

I used the following code in the snapshot to get the quaternion from OpenVins:

[image]

And it matches the ROS output so I assume it's correct displayed but output angular position to default quat: scalar:0.477802, vector:(-0.518875, 0.504593, -0.497856)  not correct after run quat must be set to scalar: 1.0, vector:(0, 0, 0) or closest to it value,  to get identity matrix3x3. This may require changing the order or sign from the axis imu measurements transmitted from the device (angular velocity & acceleration XYZ), however this will likely require re-calibration imu.

```           
    QQuaternion test (state_plus(3) /* W */ , state_plus(0) /* X */ , state_plus(1) /* Y */ , state_plus(2) /* Z */ );

     qDebug()<<test;

     QMatrix3x3 test2 = test.toRotationMatrix();

     qDebug()<<test2;

     qDebug()<<test.toEulerAngles();
                                                                                  
     Eigen::Quaternion<float, Eigen::AutoAlign> quat (state_plus(3) /* W */ , state_plus(0) /* X */ , state_plus(1) /* Y */ , state_plus(2) /* Z */ );

      Eigen::Matrix<float, 3, 3> matrix = quat.toRotationMatrix();

      matrix3x3 = matrix.data();

      qDebug()<<"  "<< matrix3x3[0]<<"  "<<matrix3x3[3]<<"  "<<matrix3x3[6];
      qDebug()<<"  "<< matrix3x3[1]<<"  "<<matrix3x3[4]<<"  "<<matrix3x3[7];
      qDebug()<<"  "<< matrix3x3[2]<<"  "<<matrix3x3[5]<<"  "<<matrix3x3[8];
```

goldbattle, 2023-11-01:
It won't be zero unless the roll and pitch of the platform is zero.
The first orientation is already the position of the IMU in the global, so
unless your IMU is perfectly aligned with gravity then this should be
non-zero.

If you are having trouble with the quaternion conversion, then use our
quat2Rot() function, and then compare using SO3.

closed 2023-11-04 by Userpc1010