rpng/open_vins #400: Setting up D435i for OpenVins

Genozen, 2023-11-18:
Hello. First, thank you for maintaining this repo and actively helping out everyone.

I'm a newer learner for using ROS 2 and OpenVins...

I would like to run real-time RealSense D435i Stereo + IMU with the OpenVins SLAM. And I have read through most of the issue pages.

I'm still a little lost on how to get OpenVins to subscribe to my camera topics correctly...

This is the subscribe node from msckf:
```
ros2 node info /ov_msckf/run_subscribe_msckf 
/ov_msckf/run_subscribe_msckf
  Subscribers:
    /cam0/image_raw: sensor_msgs/msg/Image
    /cam1/image_raw: sensor_msgs/msg/Image
    /imu0: sensor_msgs/msg/Imu
    /parameter_events: rcl_interfaces/msg/ParameterEvent
```

This is the topics from D435i:
```
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/color/points
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/camera/camera/extrinsics/depth_to_depth
```

Do I just need to do some sort of remapping in the launch file?

Or I should really do the Kalibr tutorial first, which generates a bunch of configs, and this will then be automatically configured in the subscriber launch? Just looking for more step by step details...

Genozen, 2023-11-18:
Also, side note:

Just for fun, I tried using the rs_d455 config, but had an error. The rs_t265 launches with no problem.
```
ros2 launch ov_msckf subscribe.launch.py config:=rs_d455
[INFO] [launch]: All log files can be found below /home/ascent/.ros/log/2023-11-17-22-23-19-318862-ascent-128932
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [run_subscribe_msckf-1]: process started with pid [128933]
[INFO] [rviz2-2]: process started with pid [128935]
[rviz2-2] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[run_subscribe_msckf-1] overriding node verbosity with value from ROS!
[run_subscribe_msckf-1] Setting printing level to: INFO
[ERROR] [run_subscribe_msckf-1]: process has died [pid 128933, exit code -11, cmd '/home/ascent/Desktop/workspace/install/ov_msckf/lib/ov_msckf/run_subscribe_msckf --ros-args -r __ns:=/ov_msckf --params-file /tmp/launch_params_vndi1kkb --params-file /tmp/launch_params_dn1ne2d0 --params-file /tmp/launch_params_nsfuf5w7 --params-file /tmp/launch_params_uakjeapj --params-file /tmp/launch_params_9lk0ftz9']
```

goldbattle, 2023-11-22:
You need to specify only to use a single camera instead of the default two.
The default config is for just running monocular on the RGB camera of the D455.

Hope this PR will help you: https://github.com/rpng/open_vins/issues/388

goldbattle, 2023-11-22:
RE; Your original question about the topics, you should be able to use the Kalibr output files direction. You should be able to specify the topic in the camchain.yaml to just be `/camera/camera/color/image_raw` directly (but you really should be able to directly use the one from Kalibr).

You might try following this walk through video: https://youtu.be/rBT5O5TEOV4?si=h8XTEXlX93-rwEEP&t=2107

Genozen, 2023-12-04:
Going through the Kalibr tutorial works! I will probably post an overall setup with ROS2. Thank you

closed 2023-12-07 by goldbattle

Genozen, 2024-08-09:
@marcoleg @ShreyasKuntnal  for any drift-related issue, I would first check your calibration results. (See OpenVins's guide). That's usually the number one cost I've found for drifting. The second possibility could be either the environment or you are running on Jetson (ARM). I experienced some drift using Jetson and have not investigate this closely yet.

ShreyasKuntnal, 2024-08-13:
@Genozen I have shared all the files like calibrations files on this particular issue: https://github.com/rpng/open_vins/issues/458#issuecomment-2255837093 
I will be sharing it again for your reference: [Google Drive Link](https://drive.google.com/drive/folders/1hR_MxBLU5vu8E_1GEvaO-zFruonA93Nr?usp=sharing )

I request you to sort issues occurring due to the use of the d435i camera as soon as possible, As many people have already raised the issue and all are still failing to achieve the performance as projected in all demos and cameras like d455

Userpc1010, 2024-08-13:
I also encountered a drift problem on intel d455 and I have 5-6 variance, calibration files from the ros calibr with the aprilgrid board but this did not help.

https://youtu.be/lIQRj7pLs2k?si=dmabdzAXtX6hlTUO

It seems to me that one of the important reasons for the drift is the inability of OpenVINS to work in a geometrically degraded (planar) environment. At the same time, a small number of points, 50-100 pcs, are highly susceptible to errors in their distribution across the camera image or perhaps the KLT/ORB detectors are highly affected by shadows and the sun and produce tracking errors.

In a geometry-rich environment, openvins worked well with calibration from eeprom d455 camera or default rs_d455 config.

https://youtu.be/LlvSzjQ40l4

I think, to solve the problem globally, need to make a tight coupling system where the [ov_plane](https://youtu.be/bec7LbYaOS8?si=0_NPfzoVjQNFCkty) module will work together with the usual OpenVINS and compensate for each other's shortcomings, but I don’t know about such plans.