rpng/open_vins #401: z axis of the ovmsckf/poseimu keeps gradually moving above along z axis after every loop is completed

Aishkrish18, 2023-11-22:
Hello @goldbattle 
I have been having this small issue where the z axis of the pose keeps increasing one level everytime i come to the same point and start again. I would like to know if this because of the odometry estimation method in itself or it depends on the camera I use? 
I am using ZED2 camera , and this has been a constant problem ,which did not affect much until now, since I am performing elevation mapping , which leads to map layering along z axis everytime i make a loop. Could you suggest me or give me any feedback regarding this? 
Just a side view of how it gradually increases in height.
[image]

Regards,
Aishwarya

Aishkrish18, 2023-12-11:
I am still stuck with this problem. 
Does any one else who have used ZED2 for openvins , did you face this problem where the zed baselink frame keeps drifitng away? Is this why, the z axis keeps moving above gradually?

Aishkrish18, 2023-12-11:
I figured this issue comes  from the zed camera. Some parameters were required to change in the zed 's common.yaml. 
thanks :)

goldbattle, 2023-12-12:
Do you have what exactly you changed? Others might be interested in what is
needed to run on it.

akkoyunlu, 2025-01-17:
Hİ! I have same problem. Could you find parameters that must be changed for zed in common file  ?