rpng/open_vins #403: getting error while ov_msckf package is building

omeredemen, 2023-11-29:
hello,

ubuntu 22.04 
6.2.0-37-generic
ROS2 humble

 in installation step, after this command `colcon build --event-handlers console_cohesion+ --packages-select ov_core ov_init ov_msckf ov_eval # ROS2 with verbose output
`
While ov_msckf was building the computer was freezing and after couple of tries I got the following output

[Processing: ov_msckf]                             
[Processing: ov_msckf]                                     
[Processing: ov_msckf]                                       
--- output: ov_msckf                                         
[  4%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/sim/Simulator.cpp.o
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/core/VioManagerHelper.cpp.o
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/state/Propagator.cpp.o
[ 22%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/core/VioManager.cpp.o
[ 22%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/state/StateHelper.cpp.o
[ 27%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterHelper.cpp.o
[ 45%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterSLAM.cpp.o
[ 45%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/ros/ROSVisualizerHelper.cpp.o
[ 45%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterMSCKF.cpp.o
[ 45%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterZeroVelocity.cpp.o
[ 50%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/ros/ROS2Visualizer.cpp.o
In file included from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROSVisualizerHelper.h:37,
                 from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROSVisualizerHelper.cpp:22:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.h:45,
                 from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.cpp:22:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.h:28,
                 from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.cpp:22:
/opt/ros/humble/include/image_transport/image_transport/image_transport.h:32:89: note: ‘#pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead’
   32 | #pragma message ("Warning: This header is deprecated. Use 'image_transport.hpp' instead")
      |                                                                                         ^
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_msckf_lib.dir/build.make:202: CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterSLAM.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_msckf_lib.dir/build.make:146: CMakeFiles/ov_msckf_lib.dir/src/core/VioManager.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/ov_msckf_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
--- stderr: ov_msckf
In file included from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROSVisualizerHelper.h:37,
                 from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROSVisualizerHelper.cpp:22:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.h:45,
                 from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.cpp:22:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.h:28,
                 from /home/omer/Desktop/source_code/vslam_ros2/src/open_vins/ov_msckf/src/ros/ROS2Visualizer.cpp:22:
/opt/ros/humble/include/image_transport/image_transport/image_transport.h:32:89: note: ‘#pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead’
   32 | #pragma message ("Warning: This header is deprecated. Use 'image_transport.hpp' instead")
      |                                                                                         ^
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_msckf_lib.dir/build.make:202: CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterSLAM.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_msckf_lib.dir/build.make:146: CMakeFiles/ov_msckf_lib.dir/src/core/VioManager.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/ov_msckf_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ov_msckf [1min 55s, exited with code 2]

Summary: 3 packages finished [1min 58s]
  1 package failed: ov_msckf
  1 package had stderr output: ov_msckf
I ve change some lines in /open_vins/ov_msckf/src/ros/ROS2Visualizer.h 
28 #include <image_transport/image_transport.hpp>
45 #include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>

and open_vins/ov_msckf/src/ros/ROSVisualizerHelper.h
37 #include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>

after these changes the problem was solved

goldbattle, 2023-11-29:
Likely you ran out of memory as cc1plus killed itself. Try building again,
or each package by itself (e.g. ov_core first, then ov_msckf).

omeredemen, 2023-11-30:
Thanks for the reply
I tried many times actually even I tried to build packages by one by but It kept up giving this error. But I solved this issue by changing some lines that I mentioned above

goldbattle, 2023-12-01:
What exactly did you change to get it to compile?
What you have posted are just warnings and won't prevent compilation.

omeredemen, 2023-12-05:
Yeah it shouldn't have prevent compilation. As I said, I tried many times and I got same result before making changes. Here exactly what I  changed:
[image]

and

[image]

omeredemen, 2024-04-17:
I got same issue again and this time changing these lines didn't work. So i disable grapical.target to do that
`sudo systemctl set-default multi-user.target # after that restart your system it works for me jetson nano`

or you can disable your graphic with press ctrl+alt+F3 and go to tty terminal (or F2 you can search what it is for your system) and try to build. it works for me in pc. To exit from tty terminal press  ctrl+alt+F2 (or F3)