rpng/open_vins #404: Why do transpose on rotation in B-spline curve feed_trajectory()

Alex-Beh, 2023-12-01:
Hi, I have one question about the feed_trajectory in B-spline curve. The input is the trajectory in TUM format, each line represent the translation & rotation from IMU to the global frame. However, in the [feed_trajectory()](https://github.com/rpng/open_vins/blob/e84c8ca164c1ba59b4d0cae8da66141fc6df22fe/ov_core/src/sim/BsplineSE3.cpp#L26) why you do the transpose on the rotation matrix and the translation remain the same?

https://github.com/rpng/open_vins/blob/e84c8ca164c1ba59b4d0cae8da66141fc6df22fe/ov_core/src/sim/BsplineSE3.cpp#L42-L43

Alex-Beh, 2023-12-01:
After I dive deeper into the code, I realized the function `quat_2_Rot()` is converting inverse() operation. and I guess that's why you doing `transpose()` after getting the result from `quat_2_Rot()`. May I know why need to do the inverse() operation in `quat_2_Rot()`? Same thing happened in `rot_2_quat()`

goldbattle, 2023-12-01:
The rotation is read into a JPL quaternion, so its direction flip (e.g. a
ItoG rotation in hamilton becomes a GtoI in JPL).
Thus, we transpose so that the SE3 matrix we are constructing actually
corresponds to the correct T_ItoG transformation with R_ItoG and p_GinI
values.

This is the same reason why when we internally estimate rotation R_GtoI /
q_GtoI, but when we save and external tools process it it will be the
expected R_ItoG with p_GinI values.
Hope this helps.

closed 2023-12-07 by goldbattle