rpng/open_vins #408: Suggestion for building on top of OpenVins for full mapping

Genozen, 2023-12-13:
Hello, I've recently got my d435i to work well with OpenVins and the Odom path is reasonably good.

I would like to start building a map and perform loop-closures and am wondering if you have any suggestions.

I might get a dToF (3D LiDar) so I can have 3D point clouds generated for full mapping generation.

I was thinking of sending the topics to Rtabmap in ROS 2, but this seems like a hurdle in addition to understand the code in rtabmap in addition to OpenVins... (still learning).

Any suggestions is appreciated!

goldbattle, 2023-12-31:
I think a secondary pose graph which keeps tracks of features in each keyframe would be a good start (see [ov_secondary](https://github.com/rpng/ov_secondary) as an example). Otherwise, if you could integrate as an initial guess into rtabmap that could be an interesting direction too.

borongyuan, 2024-01-14:
I have updated OpenVINS support for RTAB-Map. You just need to rebuild RTAB-Map with OpenVINS and then set Odom/Strategy to 10. In ROS you can also use OpenVINS through rtabmap_odom instead of using OpenVINS Node directly.

closed 2024-01-16 by goldbattle

borongyuan, 2024-01-18:
> Could you point me to the relevant repositories?

PRs are already merged.
https://github.com/introlab/rtabmap