rpng/open_vins #410: OpenCV memory insufficient error in ROS2 Foxy

Genozen, 2023-12-15:
Hello, I've gotten the euroc_mav data to run on OpenVins in ROS2 Humble, but when I tried it on Foxy version (tried to follow the same installation process), I get an error:

```
ros2 launch ov_msckf subscribe.launch.py config:=euroc_mav
[INFO] [launch]: All log files can be found below /home/genozen/.ros/log/2023-12-15-15-17-15-830002-genozenPC-16329
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [run_subscribe_msckf-1]: process started with pid [16331]
[INFO] [rviz2-2]: process started with pid [16333]
[run_subscribe_msckf-1] overriding node verbosity with value from ROS!
[run_subscribe_msckf-1] Setting printing level to: INFO
[run_subscribe_msckf-1] terminate called after throwing an instance of 'cv::Exception'
[run_subscribe_msckf-1]   what():  OpenCV(4.8.0-dev) /home/genozen/Documents/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 60384380995536 bytes in function 'OutOfMemoryError'
[run_subscribe_msckf-1] 
[ERROR] [run_subscribe_msckf-1]: process has died [pid 16331, exit code -6, cmd '/home/genozen/Documents/Openvins/catkin_ws_ov/install/ov_msckf/lib/ov_msckf/run_subscribe_msckf --ros-args -r __ns:=/ov_msckf --params-file /tmp/launch_params_xiywmr35 --params-file /tmp/launch_params_yfi2ibrn --params-file /tmp/launch_params_2ezbc9s7 --params-file /tmp/launch_params_oya1ci03 --params-file /tmp/launch_params_iy1wvvzk'].

```

The allocation memory looks like a memory leak? Also, checked the OpenCV version I have is 4.8.1 instead of the 4.8.0-dev, would this matter?

goldbattle, 2023-12-31:
That sure is a new error! Could you test an older opencv version if possible?

Doja-Ant, 2024-01-07:
> That sure is a new error! Could you test an older opencv version if possible?

I got the same error with opencv4.2 but in ROS noetic.

Genozen, 2024-02-09:
I will give it a try again later, humble has been working great...so just using that for now

wht102345, 2024-03-24:
Hello! I may have encountered the same problem with ubuntu20.04 ROS noetic

:~/workspace/catkin_ws_ov$ roslaunch ov_msckf subscribe.launch config:=euroc_mav
... logging to /home/wht/.ros/log/b9bfbab2-e9b7-11ee-9bd8-358f1748886e/roslaunch-wht-ubuntu-6441.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wht-ubuntu:43359/

SUMMARY
========

CLEAR PARAMETERS
 * /ov_msckf/

PARAMETERS
 * /ov_msckf/config_path: /home/wht/workspa...
 * /ov_msckf/max_cameras: 2
 * /ov_msckf/record_timing_filepath: /tmp/traj_timing.txt
 * /ov_msckf/record_timing_information: False
 * /ov_msckf/use_stereo: True
 * /ov_msckf/verbosity: INFO
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    ov_msckf (ov_msckf/run_subscribe_msckf)

ROS_MASTER_URI=http://localhost:11311

process[ov_msckf-1]: started with pid [6456]
overriding node verbosity with value from ROS!
Setting printing level to: INFO
overriding node max_cameras with value from ROS!
overriding node max_cameras with value from ROS!
overriding node use_stereo with value from ROS!
overriding node record_timing_information with value from ROS!
overriding node record_timing_filepath with value from ROS!
overriding node use_stereo with value from ROS!
subscribing to IMU: /imu0
subscribing to cam (stereo): /cam0/image_raw
subscribing to cam (stereo): /cam1/image_raw
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/core/src/matrix.cpp:235: error: (-215:Assertion failed) s >= 0 in function 'setSize'

================================================================================REQUIRED process [ov_msckf-1] has died!
process has died [pid 6456, exit code -6, cmd /home/wht/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_msckf __name:=ov_msckf __log:=/home/wht/.ros/log/b9bfbab2-e9b7-11ee-9bd8-358f1748886e/ov_msckf-1.log].
log file: /home/wht/.ros/log/b9bfbab2-e9b7-11ee-9bd8-358f1748886e/ov_msckf-1*.log
Initiating shutdown!
================================================================================
[ov_msckf-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

wht102345, 2024-03-24:
Hello! I may have encountered the same problem with ubuntu20.04 ROS noetic
[image]

goldbattle, 2024-03-25:
@wht102345 it looks like yours is a different error with the `setSize` function. Did you install opencv correctly? What version are you using?

wht102345, 2024-03-25:
Thank you very much for your answer! The picture is the version of opencv I have queried this is the version that comes with ROS noetic, but due to the previous configuration of ORB-SLAM3 need a higher version of opencv there should be another opencv on the computer, but even if I specify in Cmakelist to use opencv4.2 it still does not work properly, when I run roslaunch ov_msckf subscribe.launch when I haven't run the euroc packet it's fine, when I run the packet ~/data/Dataset/EuRoc$ rosbag play V1_01_easy.bag it randomly has the three problems in the picture. I'm a newbie and I'm at a loss as to what to do.

wht102345, 2024-03-27:
> 看起来您的错误是该函数的不同错误。你正确安装了opencv吗？您使用的是什么版本？`setSize`

Sorry, it's my mistake, my opencv version is conflicting.

wht102345, 2024-03-27:
> @wht102345 it looks like yours is a different error with the `setSize` function. Did you install opencv correctly? What version are you using?

Sorry, it's my mistake, my opencv version is conflicting.

ruichaowang, 2024-05-23:
> > @wht102345 it looks like yours is a different error with the `setSize` function. Did you install opencv correctly? What version are you using?
> 
> Sorry, it's my mistake, my opencv version is conflicting.

Hey, I'm encounted the same setsize error. 
Both the ubuntu and ros i'm using is opencv 4.2.  
Could you please explain "opencv version is conflcting" means? 
How do you solve this problem?

Thanks~