rpng/open_vins #412: Bad image quality in rviz

JackCaster, 2024-01-08:
I am streaming 800x600 frames @30 fps. I performed the camera-IMU calibration fine, but when I run `roslaunch ov_msckf subscribe.launch config_path:=<...>/openvins/estimator_config.yaml max_cameras:=1 use_stereo:=false` the image quality in Rviz is terrible. Most frames are corrupted so that the algorithm cannot work properly
[image] 
When I play a recorded bag or stream the data directly into rviz (tracking the main topic that carries uncompressed images), the video looks ok, with some sporadic corrupted frames here and there.

Do you know what could cause this issue? Do you have any suggestions on how to improve the image quality?

goldbattle, 2024-01-16:
Seems very weird, maybe there is an issue in the conversion to mono or something?
If you could pose a bag that has this problem (and config files) that would be most helpful.

JackCaster, 2024-01-16:
I uploaded the bag and config files [here](https://drive.google.com/drive/folders/1EBWJFoBLK8ZNcUrtqhkBG2DlzZM40SkT?usp=sharing). I hope there is everything you need.

I think the camera / imu calibration params may be a bit off but I am mostly interested in getting the video stream to work now. While recording the bag, I had an rqt_image_viewer window on the side and the video looked good throughout. I also replayed the bag in foxglove and the video is fine also there. But when running the real time openvins routine, the video stream falls apart. Note, the video is already grayscale. I thought that would help the processing but it did not unfortunately. 

PS: Thank you for posting those tutorials on Yoututbe---they helped immensely to get started with Kalibr and OpenVINS.

JackCaster, 2024-01-29:
Hey I was just wondering if anybody had a change of looking at the data I uploaded. I am really curious to get this algo to work!