rpng/open_vins #414: Dual stereo camera setup

ch-hms, 2024-01-18:
Hi, 

Thanks a lot for the truly amazing work. 

I would like to run OpenVINS with a sensor rig made of 2 stereo cameras (Luxonis OAK D Pro W), one pointing forward and the other one pointing backward (so there is overlap between cameras of a given stereo pair, but not between stereo pairs). Both will be connected via USB-C so, to the best of my knowledge, precise hardware time-sync will not be possible. Is it possible to run OpenVINS on this setup with the current set of features? If so, would it be possible to have some indications? I found out about the previous issues and the 'stereo_pairs' parameter, but it seems like this was removed. 

Thanks again,

Charles

goldbattle, 2024-01-21:
If you are able to get accuracy calibration to the IMU for each camera, you should be able to run with the use_stereo set to false.
It will just do monocular tracking of each camera in this case.

Genozen, 2024-02-09:
Interesting topic! So if I have 4 sets of D435i individually calibrated and have them face North, South, East, West. Would OpenVins just take each monocular tracking from each camera?

goldbattle, 2024-02-12:
Correct. There is small issue with you would need to implement this subscriber as we only support multiple *synchronized*. If you want to do async cameras (with different time offsets per-camera and triggered asynchronously), then you might want to checkout MINS here: https://github.com/rpng/MINS

closed 2024-02-23 by goldbattle