rpng/open_vins #420: Why abandon the state and features estimated during initialization?

TianzeXie, 2024-02-20:
I notice that you left two variables in InertialInitializer.cpp
[137](https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_init/src/init/InertialInitializer.cpp#L137)
[138](https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_init/src/init/InertialInitializer.cpp#L138)
which should be used to propagate the imu states and features from initialization to filtering. But they are not be used.
I think it will be more robust and accurate with them, although calculate them would cause a lot of delay. But after some experiment it gives some negative effect to the filtering. WHY?
@goldbattle

goldbattle, 2024-02-23:
You can give it a try, but one challenge is that the covariance of the complete system needs to be recovered. If the features and clones are also recovered this state size will be large, and thus the covariance recovery will become large (it is O(N^3) process). We only do the 15x15 covariance to try to recover the covariance as quickly as possible.

You might need to do some inflation as I have found that that covariance typically recovered is overconfident due to the very limited motion over the short period. There is a [simulator](https://github.com/rpng/open_vins/tree/master/ov_init/src/sim), so you should be able to play around for yourself.

TianzeXie, 2024-02-23:
Thanks for your reply! 
I have tried it and found the covariance issue. In my experiment, it performed more accurate under 0.5 bag rate. To deal with such trade-off problem, I tried to recover the latest n state, sometimes it truly perform better.

closed 2024-02-23 by goldbattle