rpng/open_vins #421: OpenVins on Jetson Orin Nano?

Genozen, 2024-02-20:
Hello, I'm trying to build the OpenVins ROS 2 on Jetson Orin Nano... I did see some old post stating that it works on the older Jetson, so I wanted to confirm if it is runnable.

While I tried to follow the build instructions, I already ran into `Aruco` related issue, is it because my OpenCV is too new?

```
colcon build --event-handlers console_cohesion+ --packages-select ov_core ov_init ov_msckf ov_eval # ROS2 with verbose output
Starting >>> ov_core 
--- output: ov_core                            
-- OPENCV: 4.8.0 | BOOST: 1.74.0
-- PYTHON VERSION: 3.10.12
-- PYTHON INCLUDE: /usr/include/python3.10
-- PYTHON LIBRARIES: /usr/lib/aarch64-linux-gnu/libpython3.10.so
-- ROS *2* version found, building ROS2.cmake
-- Found rclcpp: 16.0.8 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.6 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found cv_bridge: 3.2.1 (/opt/ros/humble/share/cv_bridge/cmake)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/genozen/Documents/ros2_openvins/catkin_ws_ov/build/ov_core
Consolidate compiler generated dependencies of target ov_core_lib
[  5%] Building CXX object CMakeFiles/ov_core_lib.dir/src/track/TrackAruco.cpp.o
In file included from /home/genozen/Documents/ros2_openvins/catkin_ws_ov/src/open_vins/ov_core/src/track/TrackAruco.cpp:22:
/home/genozen/Documents/ros2_openvins/catkin_ws_ov/src/open_vins/ov_core/src/track/TrackAruco.h:26:10: fatal error: opencv2/aruco.hpp: No such file or directory
   26 | #include <opencv2/aruco.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_core_lib.dir/build.make:146: CMakeFiles/ov_core_lib.dir/src/track/TrackAruco.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/ov_core_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
--- stderr: ov_core
In file included from /home/genozen/Documents/ros2_openvins/catkin_ws_ov/src/open_vins/ov_core/src/track/TrackAruco.cpp:22:
/home/genozen/Documents/ros2_openvins/catkin_ws_ov/src/open_vins/ov_core/src/track/TrackAruco.h:26:10: fatal error: opencv2/aruco.hpp: No such file or directory
   26 | #include <opencv2/aruco.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_core_lib.dir/build.make:146: CMakeFiles/ov_core_lib.dir/src/track/TrackAruco.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/ov_core_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ov_core [2.31s, exited with code 2]

Summary: 0 packages finished [
```

Genozen, 2024-02-20:
Bypassed the `#include <opencv2/aruco.hpp>` by commenting it out and built successfully.

Onto the `ov_msckf`

```
[Processing: ov_msckf]                                        
--- output: ov_msckf                                          
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: /usr (found suitable version "4.8.0", minimum required is "4") 
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: system filesystem thread date_time 
-- Found Threads: TRUE  
-- Found required Ceres dependency: Eigen version 3.4.0 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 2.0.0 installed in: /usr with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, Multithreading]
-- OPENCV: 4.8.0 | BOOST: 1.74.0 | CERES: 2.0.0
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- ROS *2* version found, building ROS2.cmake
-- Found rclcpp: 16.0.8 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.6 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "3.0.2")  
-- Found FastRTPS: /opt/ros/humble/include  
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found tf2_ros: 0.25.5 (/opt/ros/humble/share/tf2_ros/cmake)
-- Found tf2_geometry_msgs: 0.25.5 (/opt/ros/humble/share/tf2_geometry_msgs/cmake)
-- Found eigen3_cmake_module: 0.1.1 (/opt/ros/humble/share/eigen3_cmake_module/cmake)
-- Found Eigen3: TRUE (found version "3.4.0") 
-- Ensuring Eigen3 include directory is part of orocos-kdl CMake target
-- Found sensor_msgs: 4.2.3 (/opt/ros/humble/share/sensor_msgs/cmake)
-- Found nav_msgs: 4.2.3 (/opt/ros/humble/share/nav_msgs/cmake)
-- Found cv_bridge: 3.2.1 (/opt/ros/humble/share/cv_bridge/cmake)
-- Found image_transport: 3.1.8 (/opt/ros/humble/share/image_transport/cmake)
-- Found ov_core: 2.7.0 (/home/genozen/Documents/ros2_openvins/catkin_ws_ov/install/ov_core/share/ov_core/cmake)
-- Found ov_init: 2.7.0 (/home/genozen/Documents/ros2_openvins/catkin_ws_ov/install/ov_init/share/ov_init/cmake)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/genozen/Documents/ros2_openvins/catkin_ws_ov/build/ov_msckf
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/dummy.cpp.o
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/state/StateHelper.cpp.o
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/sim/Simulator.cpp.o
[ 18%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/state/Propagator.cpp.o
[ 22%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/state/State.cpp.o
[ 27%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/core/VioManager.cpp.o
[ 31%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/core/VioManagerHelper.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```

Genozen, 2024-02-20:
Full message:

```
colcon build --event-handlers console_cohesion+ --packages-select ov_msckf # ROS2 with verbose output
Starting >>> ov_msckf
[Processing: ov_msckf]                             
[Processing: ov_msckf]                                 
[Processing: ov_msckf]                                       
[Processing: ov_msckf]                                       
[Processing: ov_msckf]lete] [ov_msckf:build 27% - 2min 30.9s]
[Processing: ov_msckf]                                       
--- output: ov_msckf                                         
Consolidate compiler generated dependencies of target ov_msckf_lib
[  4%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/sim/Simulator.cpp.o
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterSLAM.cpp.o
[ 13%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterHelper.cpp.o
[ 18%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterZeroVelocity.cpp.o
[ 27%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterMSCKF.cpp.o
[ 27%] Building CXX object CMakeFiles/ov_msckf_lib.dir/src/core/VioManagerHelper.cpp.o
In file included from /usr/include/boost/math/distributions/chi_squared.hpp:13,
                 from /home/genozen/Documents/ros2_openvins/catkin_ws_ov/src/open_vins/ov_msckf/src/update/UpdaterMSCKF.cpp:36:
/usr/include/boost/math/special_functions/gamma.hpp: In instantiation of ‘boost::math::detail::upper_incomplete_gamma_fract<T>::result_type boost::math::detail::upper_incomplete_gamma_fract<T>::operator()() [with T = long double; boost::math::detail::upper_incomplete_gamma_fract<T>::result_type = std::pair<long double, long double>]’:
/usr/include/boost/math/tools/fraction.hpp:216:20:   required from ‘typename boost::math::tools::detail::fraction_traits<Gen>::result_type boost::math::tools::continued_fraction_a(Gen&, const U&, uintmax_t&) [with Gen = boost::math::detail::upper_incomplete_gamma_fract<long double>; U = long double; typename boost::math::tools::detail::fraction_traits<Gen>::result_type = long double; uintmax_t = long unsigned int]’
/usr/include/boost/math/tools/fraction.hpp:251:31:   required from ‘typename boost::math::tools::detail::fraction_traits<Gen>::result_type boost::math::tools::continued_fraction_a(Gen&, const U&) [with Gen = boost::math::detail::upper_incomplete_gamma_fract<long double>; U = long double; typename boost::math::tools::detail::fraction_traits<Gen>::result_type = long double]’
/usr/include/boost/math/special_functions/gamma.hpp:317:68:   required from ‘T boost::math::detail::upper_gamma_fraction(T, T, T) [with T = long double]’
/usr/include/boost/math/special_functions/gamma.hpp:1157:44:   required from ‘T boost::math::detail::gamma_incomplete_imp(T, T, bool, bool, const Policy&, T*) [with T = long double; Policy = boost::math::policies::policy<boost::math::policies::promote_float<false>, boost::math::policies::promote_double<false>, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy>]’
/usr/include/boost/math/special_functions/detail/igamma_inverse.hpp:356:67:   required from ‘std::tuple<T, T, T> boost::math::detail::gamma_p_inverse_func<T, Policy>::operator()(const T&) const [with T = double; Policy = boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy>]’
/usr/include/boost/math/tools/roots.hpp:560:32:   required from ‘T boost::math::tools::detail::second_order_root_finder(F, T, T, T, int, uintmax_t&) [with Stepper = boost::math::tools::detail::halley_step; F = boost::math::detail::gamma_p_inverse_func<double, boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >; T = double; uintmax_t = long unsigned int]’
/usr/include/boost/math/tools/roots.hpp:710:64:   required from ‘T boost::math::tools::halley_iterate(F, T, T, T, int, uintmax_t&) [with F = boost::math::detail::gamma_p_inverse_func<double, boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >; T = double; uintmax_t = long unsigned int]’
[... cut ...]
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ov_msckf_lib.dir/build.make:216: CMakeFiles/ov_msckf_lib.dir/src/update/UpdaterZeroVelocity.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/ov_msckf_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ov_msckf [3min 12s, exited with code 2]

Summary: 0 packages finished [3min 14s]
  1 package failed: ov_msckf
  1 package had stderr output: ov_msckf

```

Genozen, 2024-02-21:
Problem resolved with using less RAM to build

closed 2024-02-21 by Genozen

Genozen, 2024-02-21:
@goldbattle , hello I have a quick question regarding OpenVins on Jetson Orin (8 gb ram).

I observed some noticeable drift occurred when running the standard `Euroc_mav` config with the tutorial ROS 2 bag (V1_Easy.bag). The output rate looks normal, but compared to my setup with the Laptop, the pose estimation significantly drifts easily...

Any idea as to why this might be? Did skipping the `Aruco.hpp` build make any difference?

goldbattle, 2024-02-23:
Hi, to build on embedded devices you should add some swap memory, or you can sometimes build with `-j1` (single thread) to try to reduce the memory use while building. I you figured out that `Killed signal terminated program cc1plus` means the device ran out of memory.

I am not sure of any big differences on why there would be significant trajectory drift. Disabling the aruco should be no problem.

Set `-DENABLE_ARUCO_TAGS=OFF` to disable the aruco compile. You can also see the cmake flags you can set (not sure how you can pass them in ROS2).: https://docs.openvins.com/gs-installing-free.html

Genozen, 2024-02-28:
If anyone wants to build the Aruco:

I've added:
```
#include "opencv2/objdetect/aruco_detector.hpp"
#include "opencv2/aruco/aruco_calib.hpp"
```

instead of just
`#include <opencv2/aruco.hpp>` in the `ov_core`

This then builds correctly without disabling the aruco tag.

fuwafuwaboom, 2024-03-10:
> If anyone wants to build the Aruco:
> 
> I've added:
> 
> ```
> #include "opencv2/objdetect/aruco_detector.hpp"
> #include "opencv2/aruco/aruco_calib.hpp"
> ```
> 
> instead of just `#include <opencv2/aruco.hpp>` in the `ov_core`
> 
> This then builds correctly without disabling the aruco tag.

Hi Genozen, Same problem when building it on Jetson. So grateful for your method, it can solve the bug of " opencv2/aruco.hpp: No such file or directory" but there are still other bugs of aruco like:  "opencv2/objdetect/aruco_detector.hpp: No such file or directory" 

I wondering if there is a method like install some package instead of modify the .h file one by one if we have to use the auroc.

Genozen, 2024-08-09:
> > If anyone wants to build the Aruco:
> > I've added:
> > ```
> > #include "opencv2/objdetect/aruco_detector.hpp"
> > #include "opencv2/aruco/aruco_calib.hpp"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > instead of just `#include <opencv2/aruco.hpp>` in the `ov_core`
> > This then builds correctly without disabling the aruco tag.
> 
> Hi Genozen, Same problem when building it on Jetson. So grateful for your method, it can solve the bug of " opencv2/aruco.hpp: No such file or directory" but there are still other bugs of aruco like: "opencv2/objdetect/aruco_detector.hpp: No such file or directory"
> 
> I wondering if there is a method like installing some packages instead of modifying the .h file one by one if we have to use the auroc.

I didn't look too deep into it since I just need to patch it and validate it. I would assume you'd have to change the CMake portion internally to match up...  if you need the Aruco portion. Otherwise, if you don't need it, you can try what goldbattle suggested above too.

2b-t, 2025-02-21:
I ran into the same problem on an Nvidia Jetson AGX Orin running Nvidia Jetpack 6.0 as well as ROS 2. It seems that the OpenCV which comes with ROS 2 for the Jetson does not come with the Aruco module. Installing the corresponding modules with [`$ sudo apt-get install libopencv-contrib-dev`](https://github.com/opencv/opencv_contrib) would uninstall the standard OpenCV as well as several ROS 2 Debian packages related to image processing.

I proceeded to **build OpenCV from source** running the [following scripts](https://github.com/AastaNV/JEP/tree/master/script), in my case [`install_opencv4.9.0_Jetpack6.0.sh`](https://github.com/AastaNV/JEP/blob/911e9f9bf905589e056819cfc4ff49e7be4bec0b/script/install_opencv4.9.0_Jetpack6.0.sh), and then **force Colcon to link against that version of OpenCV**:

`$ colcon build --cmake-args -DOpenCV_DIR=/usr/local/lib/cmake/opencv4`

(ideally make that persistent by adding this to `~/.colcon/defaults.yaml` as described [here](https://colcon.readthedocs.io/en/released/user/configuration.html#defaults-yaml)) as well as adding

`export LD_LIBRARY_PATH=/usr/local/lib/:${LD_LIBRARY_PATH}`

to my `~/.bashrc`.