rpng/open_vins #422: Implementation using a monocular camera

lfescobarc, 2024-02-21:
I am trying to use OpenVins to SLAM with a drone; I have some restrictions; the most important one here is that I can only carry on a monocular proprietary camera, so I think OpenVins is helpful. I am trying to find a guide on using this in my case, but I have not seen it here.

Genozen, 2024-02-22:
Hi, I was able to get some cameras up and running with OpenVins following their official website. Patrick does a good job in explaining it: https://docs.openvins.com/gs-calibration.html

goldbattle, 2024-02-23:
We are unable to run with just a single monocular camera, but if you also have an IMU on the platform you should be able to run if you can calibrate the system. You should take a look at the getting started guide to get familiar with the system: https://docs.openvins.com/gs-tutorial.html

Likely you will be interested with the `max_cameras` which you can set to `1` to run with monocular mode. I recommend testing on some example datasets first to understand what the performance is like. Hope this helps.