rpng/open_vins #423: ROS2 and OpenVINS datasets

Colerock, 2024-02-21:
I have ROS2, the latest official "iron" distro, skipped ROS1 for now since I am on ubuntu 22.x, but I might downgrade later. if I cant get proper help here.

I have successfully run OpenVINS with MSCKF using the simple tutorial and the default euroc_mav dataset which is in db3 format, I am trying to run a dataset from UZH-FPV but the dataset are not recorded as .db3 and we can only download either a bag file (no metadata file included) or raw files in terms of txt files and of course images that can be download, these can be converted, but is a major hassle to do so. 

Can you explain how you managed to run the UZH-FPV dataset using OpenVINS and what I need to do in which file to run for for example "indoor forward_3_davis_gt" with ROS2. The documentation is very unclear for several datasets especially using ROS2. For the simple tutorial with ROS1 and the .db3 euroc_mav dataset is was pretty straight forward.

lfescobarc, 2024-02-22:
I am running using ROS2 as well, I am having troubles with setting a realsense d457 to work, I was able to do it with a T265, do you have any insight on how to set up the D457 camera?

goldbattle, 2024-02-23:
Hi, I converted the [eurocmav](https://docs.openvins.com/gs-datasets.html#gs-data-euroc) datasets using the method described on this page:
https://docs.openvins.com/dev-ros1-to-ros2.html

I can highly recommend the [rosbags](https://gitlab.com/ternaris/rosbags) python tool which should work for you out of the box to convert datasets into the ROS2 format for you to test with. I think all datasets are ROS1, so you will need to do this conversion, but if you have the ROS1 bags, then this should be straight forward.

```
pip3 install rosbags>=0.9.11
rosbags-convert V1_01_easy.bag --dst <ros2_bag_folder>
```