rpng/open_vins #426: ROS Free Complie and Run using WebCam

mh-ih, 2024-02-27:
I have built and make open_vins on ubuntu 20.04.6 using the installation guide (ROS Free). But now I don't know how to compile and run it using my webcam without ROS. Could anyone please help me?!
Thanks.

Userpc1010, 2024-02-28:
You need to install it as a regular library to begin with, but if you have ROS installed it will probably install the ROS version. You can solve this problem like this: 

```
cmake_minimum_required(VERSION 3.3)
project(ov_msckf) #This needs to be done for each project in the OpenVINS directory

# Include libraries (if we don't have opencv 4, then fallback to opencv 3)
# The OpenCV version needs to match the one used by cv_bridge otherwise you will get a segmentation fault!
find_package(Eigen3 REQUIRED)
find_package(OpenCV 3 QUIET)
if (NOT OpenCV_FOUND)
    find_package(OpenCV 4 REQUIRED)
endif ()
find_package(Boost REQUIRED COMPONENTS system filesystem thread date_time)
find_package(Ceres REQUIRED)
message(STATUS "OPENCV: " ${OpenCV_VERSION} " | BOOST: " ${Boost_VERSION} " | CERES: " ${Ceres_VERSION})

# If we will compile with aruco support
option(ENABLE_ARUCO_TAGS "Enable or disable aruco tag (disable if no contrib modules)" ON)
if (NOT ENABLE_ARUCO_TAGS)
    add_definitions(-DENABLE_ARUCO_TAGS=0)
    message(WARNING "DISABLING ARUCOTAG TRACKING!")
else ()
    add_definitions(-DENABLE_ARUCO_TAGS=1)
endif ()

# We need c++14 for ROS2, thus just require it for everybody
# NOTE: To future self, hope this isn't an issue...
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)

# Enable compile optimizations
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -O3 -fsee -fomit-frame-pointer -fno-signed-zeros -fno-math-errno -funroll-loops")

# Enable debug flags (use if you want to debug in gdb)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -g3 -Wall -Wuninitialized -fno-omit-frame-pointer")

# Find our ROS version!  
# NOTE: Default to using the ROS1 package if both are in our enviroment
# NOTE: https://github.com/romainreignier/share_ros1_ros2_lib_demo
find_package(catkin QUIET COMPONENTS roscpp)
find_package(ament_cmake QUIET)
#if (catkin_FOUND)                                                               Comment out this code!!!!!
#    message(STATUS "ROS *1* version found, building ROS1.cmake")
#    include(${CMAKE_CURRENT_SOURCE_DIR}/cmake/ROS1.cmake)
#elseif (ament_cmake_FOUND)
#    message(STATUS "ROS *2* version found, building ROS2.cmake")
#    include(${CMAKE_CURRENT_SOURCE_DIR}/cmake/ROS2.cmake)
#else ()
    message(STATUS "No ROS versions found, building ROS1.cmake")
    include(${CMAKE_CURRENT_SOURCE_DIR}/cmake/ROS1.cmake)
#endif ()
```

This way you will have a version without ROS even if it is present in the system. Install next as a regular cmake library.. & make -j4 & make install

Connect it to your project, I do this using qmake in pro.  :

```
INCLUDEPATH += /usr/include/boost

LIBS += -lboost_system -lboost_thread -lboost_chrono -lboost_filesystem

INCLUDEPATH += /usr/local/include/eigen3

unix:!macx: LIBS += -L$$PWD/../../../../usr/local/lib/ -lov_core_lib

INCLUDEPATH += $$PWD/../../../../usr/local/include/open_vins
DEPENDPATH += $$PWD/../../../../usr/local/include/open_vins
[... 329 lines cut ...]
  if (img_history.empty()) return;

  cv::Mat result (120, 480, CV_8UC3, cv::Scalar(10, 100, 150));

  cv::resize(img_history, result, result.size(), 0, 0, cv::INTER_CUBIC);

  emit future_image(result);
 }
}

```

Create an instance of the class in a separate thread:

```
std::string config_path = "/home/sencis/build-UGV-Desktop-Debug/UGV_Core/estimator_config.yaml";

 thread_openvins = new QThread(this);
 openvins = new OpenVins(config_path, nullptr);
 openvins->moveToThread(thread_openvins);
 thread_openvins->start();
```

The current configuration works for Stereo so you just have to deal with the camera in mono mode.
You can see all this in ROS1Visualizer.cpp and run_simulation.cpp I think this can serve as a starting point.

mh-ih, 2024-03-02:
Thanks. Honestly I am new to OpenVins. I will do what you wrote and give you feedback.

mh-ih, 2024-03-03:
It seems that you are reading the images with the real sense camera but I don't have that and want to read input images with my WebCam. Can you please explain me how to do it?

Userpc1010, 2024-03-03:
Perhaps I will leave this to you as homework)). There are examples on the Internet of how to obtain images from a camera using OpenCV, and how to use mono can be found in ROS1Visualizer.cpp. You will also need a synchronized IMU along with the camera and this is a problem because... regular cameras do not have imu, and synchronization assumes that imu data should be read at the moment when the camera shutter is charging, i.e. it must be implemented in hardware between sensors to function properly. But perhaps you have a camera like this with an ExtTrigg output from the camera shutter to read this interrupt as a read command to the IMU, or just a Realsense camera that will do the job for you.

goldbattle, 2024-03-25:
Are you trying to run it with your webcam only? This is visual-inertial project so you also need an IMU. If you just wish to use the feature tracking, then you should be able to just build ov_core and use the webcam example.

```
cd ov_core
mkdir build && cd build
cmake -DENABLE_ROS=OFF ..
./test_webcam
```

closed 2024-11-23 by goldbattle