rpng/open_vins #428: Find a mistake about FEJ in the home page

TianzeXie, 2024-03-07:
in this page https://docs.openvins.com/fej.html
At the last equation ,should it be 
$$\tilde{z_{k+1}}= z_{k+1}-h(x_{k+1})=\bar{H_k}(x_{k+1}-\hat{x}_k)$$
Because every time the filter update, we assume that the robot is at the true state of the k+1 timestamp where it observe the features.

goldbattle, 2024-03-25:
The equation on the website should be correct as this is a linearization equation. We do not know the true state, so it is not possible to evaluate the equation you have posted. Thus we linearize the true equations `z = h(x) + n` to get our `z = h(x_hat) + H (x - x_hat) + n`. The key to FEJ is about what point to evaluate the Jacobian at the ensure correct observability properties. Hope this helps.

TianzeXie, 2024-03-26:
true, just like you said, the ture equation is :
$z_{k}=h(x_k)+n$. In my context, it is $z_{k+1}=h(x_{k+1})+n$, so I have my first equation: $\tilde{z_{k+1}}=z_{k+1}-h(x_{k+1})$, right? And then consider the FEJ, I linearize it at the point $\hat{x_h}$, so I get my second equation.

goldbattle, 2024-03-30:
Typically the residual is defined after linearization. The third equation you wrote is just directly equal to noise not the residual.

closed 2024-11-23 by goldbattle