rpng/open_vins #430: Feature descriptor that is used in OpenVINS

Aishkrish18, 2024-03-11:
Hello @goldbattle 
I have a conceptual question. 
Does OpenVINS use a KLT descriptor or ORB Descriptor? What is the advantage of one over the other?
Considering an algorithm like ORBSLAM3 upon intensive testing I find OpenVINS to provide much more accurate localization, even though it's a VIO method and not SLAM. I am interested to know the reason and maybe some links and papers attached.

Regards,
AIshwarya

Alex-Beh, 2024-03-18:
Hi @Aishkrish18, I am interested on this question also. May I know the LandmarkRepresentation that you used in Open-VINS?

My two cents are that, based on study and experimentation, using ANCHORED_FULL_INVERSE_DEPTH might yield better results. Additionally, the initialization and IMU preintegration differ between ORB-SLAM3 and OpenVINS.

goldbattle, 2024-03-25:
Hi, the KLT provides much longer tracks then the descriptor. When I originally implemented I believe I saw that the descriptors were only tracked for ~5 frames even with good datasets (eurocmav v101). KLT on the other hand has much longer track length which is much more useful to us since we estimate SLAM in addition to MSCKF features.

You can test with the [test_tracking](https://github.com/rpng/open_vins/blob/master/ov_core/src/test_tracking.cpp) file to see how things perform.

closed 2024-11-23 by goldbattle