rpng/open_vins #431: Cannot build the content

Jawaheru, 2024-03-13:
After i catkin_make inside Workspace/catkin_ws_ov
It proceeds very well but it freezes after start to scanning dependencies of target ov_core_lib
I interrupt it after 10 min and the white failures shows[image]

goldbattle, 2024-03-25:
Hi, I saw you edited your post. Were you able to solve the original problem? It looks like it is because you killed the building. If you are running into memory issues you can build with `catkin build -j1` to limit the number of parallel builds.

goldbattle, 2024-03-25:
I now see you opened a second issue https://github.com/rpng/open_vins/issues/432. Closing this one

closed 2024-03-25 by goldbattle

Jawaheru, 2024-03-25:
I killed the build after it took me forever waiting to complete, however I tried the command previously and its not working. I will get back to building this and update it if any further solution worked. Ty > I now see you opened a second issue https://github.com/rpng/open_vins/issues/432. Closing this one