rpng/open_vins #433: Why dont inflate the covariance of position?

TianzeXie, 2024-03-20:
Why dont inflate the covariance of position but the other state like orientation velocity and other two bias?

goldbattle, 2024-03-21:
Hi, could you expand on what exactly you are referring to? Thanks!

TianzeXie, 2024-03-22:
here:[https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_init/src/dynamic/DynamicInitializer.cpp#L1076](url)
In my opinion, the reason is the covariance of position donot impact the other states because we dont use position to calculate any other state.

goldbattle, 2024-03-25:
Hi, the global position and yaw are unobservable for visual-inertial systems, so there is no need to inflate these parameters are we treat them as 100% known. You can see this in the prior we assign (can also see we put a prior on bias as it is near unobservable for small accel and gyo changes):
https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_init/src/dynamic/DynamicInitializer.cpp#L696-L699

We just inflate the initial orientation as its roll and pitch are observable. I found in simulation that the covariance was overconfident when using short initialization windows. This is attributed to the fact that we need full 3d motion (orientation and non-constant velocity) to fully recover our states. Hope this helps.

closed 2024-07-27 by goldbattle