rpng/open_vins #435: Pose diverging on Raspberry Pi 4 using EUROC dataset

mzahana, 2024-03-27:
Hi @goldbattle 

I am trying to run open_vins on a Raspberry Pi 4 with 4GB RAM.  I installed ROS2 humble from source and it works fine.
To compile open_vins. I first installed deps
```bash
sudo apt-get update && sudo apt-get install -y \
    libeigen3-dev \
    libboost-all-dev \
    libceres-dev \
```

Then I compiled the ros workspace using
```bash
MAKEFLAGS="-j1 -l1" colcon build --executor sequential   --cmake-args -DCMAKE_BUILD_TYPE=Release
```

The compilation went fine.

My OpenCV version is `4.6.0`

I tries using the default euroc_mav config file, the ran the `ros2 launch ov_msckf subscribe.launch.py config:=euroc_mav`, but the pose diverges quickly. I played the `V1_01_easy` bag.

I updated the config file as follows
```yaml
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 2 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 25 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 20 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: false
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 1.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
[... 348 lines cut ...]
Here is the output of `ros2 run ov_core test_profile`
```bash
Setting printing level to: INFO
OPENCV: RANDOM BIG IMAGE: 21.333 +- 0.381ms
OPENCV: HISTOGRAM EQUALIZATION: 1.114 +- 0.431ms
OPENCV: BUILD OPTICAL FLOW PYRAMID: 5.240 +- 0.495ms
OPENCV: FAST FEATURE EXTRACTION: 44.130 +- 0.329 ms (90402.12 avg feats)
OPENCV: KLT OPTICAL FLOW: 24.274 +- 0.791 ms (415.51 avg feats)
EIGEN3(double): RANDOM BIG MATRIX: 13.439 +- 0.096ms
EIGEN3(double): MATRIX MULTIPLY: 15.839 +- 0.138ms
EIGEN3(double): MATRIX INVERSION: 28.371 +- 0.105ms
EIGEN3(double): HOUSEHOLDER QR FULL: 65.858 +- 0.399ms
EIGEN3(double): HOUSEHOLDER QR PIV: 272.822 +- 1.414ms
EIGEN3(double): HOUSEHOLDER QR CUSTOM: 260.112 +- 0.872ms
EIGEN3(float): RANDOM BIG MATRIX: 10.829 +- 0.212ms
EIGEN3(float): MATRIX MULTIPLY: 7.732 +- 0.034ms
EIGEN3(float): MATRIX INVERSION: 13.859 +- 0.086ms
EIGEN3(float): HOUSEHOLDER QR FULL: 215.403 +- 0.222ms
EIGEN3(float): HOUSEHOLDER QR PIV: 74.850 +- 0.736ms
EIGEN3(float): HOUSEHOLDER QR CUSTOM: 67.022 +- 0.581ms
```

Is there any optimization that can be done for the filter to work OK on Raspberry Pi 4 with 4GB RAM?

Thank you for your time.

Hohyeong, 2024-04-18:
I have the same issue on RPI 4B 4GB RAM, using ROS2 galactic. I'm using opencv 4.5.0.

mzahana, 2024-04-18:
@Hohyeong  I am not sure if it is related to the computation power of RPi 4 or the dataset, but I managed to run the estimator using real-time stereo images on the same RPi 4 and MPU6050 IMU.

Hohyeong, 2024-04-20:
@mzahana When I applied the EuroC dataset V1_01 easy to OPENvins, I checked that the state diverges. Was the state converging properly when do you use real-time sensor? As I check with other issues, I saw that ROS2 often has problems in multi-threading environment. So I'm considering about using OPENvins using ROS1.

mzahana, 2024-04-20:
Yes. It was not diverging (at least not quickly) for real time sensor. However, it didn't work properly on Euroc dataset.

I am also interested to know how it performs on ros1. Do you gave a way to setup ros1 on Raspberry Pi 4 OS?

goldbattle, 2024-07-27:
Maybe this is related to playback and reading the ROS bag from disk in realtime. Not sure if this could be a bottleneck. If you were able to get it working with live sensor data, but not the rosbag, then this would be my suspicion.