rpng/open_vins #436: drift with rpi camera v2 and mpu9250 on jetson nano

omeredemen, 2024-04-22:
hello i am trying openvins on jetson nano
camera: raspberry pi camera v2
imu: mpu9250
and the system:
[image]

I collected 3 hours imu data to calculate allan variance parameters. And here are parameters

>   accelerometer_noise_density: 0.05166104637481483
  accelerometer_random_walk: 0.00995698308
  gyroscope_noise_density: 0.1495249516640554
  gyroscope_random_walk: 0.00844715901
  rostopic: '/imu' #Make sure this is correct
  update_rate: 100.0 #Make sure this is correct

And i handled camera and camera-imu calibration with kalibr here is the calibration results: 
[results.zip](https://github.com/rpng/open_vins/files/15059477/results.zip)

I edited kalibr_imu_chain.yaml and kalibr_imucam_chain.yaml files with calibration values. Also i change some parameters in estimator_config.yaml file that i copied from  euroc_mav/estimator_config.yaml. Here is my estimator_config.yaml file: 
[estimator_config.txt](https://github.com/rpng/open_vins/files/15059459/estimator_config.txt)

when i tried for the first time it drifted like this so whenever i try with new calibrations nothing change.
[image]

goldbattle, 2024-04-22:
Wow, those IMU noises are far to big and hits that your IMU driver isn't correct. What are you using for that?

Looking at the kalibr timing it seems your IMU has bad timestamps, which would also affect your IMU noise values.
[image]

It looks like Kalibr didn't even solve correctly, so I would start by investigating your IMU and getting a good kalibr calibration first.

omeredemen, 2024-04-22:
well I couldnt manage to fix timestamp issue. i was trying to fix with software but i got this result. so now i just read the register and publish with ros in my driver 
here is my another calibration reports (driver just read imu)
[rpi-mpu-report-cam.pdf](https://github.com/rpng/open_vins/files/15059996/rpi-mpu-report-cam.pdf)
[rpi-mpu-report-imucam.pdf](https://github.com/rpng/open_vins/files/15059997/rpi-mpu-report-imucam.pdf)

still has noises and bad timestamp and i dont know how to solve the issue. should i change my imu or driver?

omeredemen, 2024-05-06:
> Looking at the kalibr timing it seems your IMU has bad timestamps, which would also affect your IMU noise values. 

I tried to reduce my imu noises with getting avarege of 500hz data and publish them as 100hz and fix bad timestamp with software (but still couldnt get a straight line)

> It looks like Kalibr didn't even solve correctly, so I would start by investigating your IMU and getting a good kalibr calibration first.

I didnt understand this part How should i investigate my IMU?

goldbattle, 2024-11-23:
Correct. If you have a fixed frequency IMU, then you really should have only +-1-2ms variance in these timestamp deltas. 20ms difference is super big.