rpng/open_vins #442: ROS2 Initialzation error

VivekMangeUD, 2024-05-02:
Hello

I am facing initialization error when testing Openvins on my Laptop Ubuntu 22. Even after moving in random direction and Jerk motions, Its not able to initialize openvins. I have attached a recording for the same. I have also shared the launch file to run the node. 
I tried changing the below 3 varaibles in the estimator config file but still same results

**init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 1.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0** 

I am also facing issue visualizing the messages on terminal even with verbosity:= DEBUG

**ros2 launch ov_msckf ros2_aqua2_subscribe.launch verbosity:=DEBUG
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [run_subscribe_msckf-1]: process started with pid [176172]**

**Launch File**

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, LogInfo, OpaqueFunction
from launch.conditions import IfCondition
from launch.substitutions import LaunchConfiguration, TextSubstitution
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory, get_package_prefix
import os
import sys

launch_args = [
    DeclareLaunchArgument(name='namespace',         default_value='',           description='namespace'),
    DeclareLaunchArgument(name='config',            default_value='aqua2_sync',  description='euroc_mav, tum_vi, rpng_aruco...'),
    DeclareLaunchArgument(name='verbosity',         default_value='DEBUG',       description='ALL, DEBUG, INFO, WARNING, ERROR, SILENT'),
    DeclareLaunchArgument(name='use_stereo',        default_value='true',       description=''),
    DeclareLaunchArgument(name='max_cameras',       default_value='2',          description='')
]

def launch_setup(context):
    configs_dir=os.path.join(get_package_share_directory('ov_msckf'),'config')
    available_configs = os.listdir(configs_dir)
    config = LaunchConfiguration('config').perform(context)
    if not config in available_configs:
        return[LogInfo(msg='ERROR: unknown config: \'{}\' - Available configs are: {} - not starting OpenVINS'.format(config,', '.join(available_configs)))]
    config_path = os.path.join(get_package_share_directory('ov_msckf'),'config',config,'estimator_config.yaml')
    node1 = Node(package = 'ov_msckf',
                 executable = 'run_subscribe_msckf',
                 namespace = LaunchConfiguration('namespace'),
                 parameters =[{'verbosity': LaunchConfiguration('verbosity')},
                              {'use_stereo': LaunchConfiguration('use_stereo')},
                              {'max_cameras': LaunchConfiguration('max_cameras')},
                              {'config_path': config_path}])
    return [node1]

def generate_launch_description():
    opfunc = OpaqueFunction(function = launch_setup)
    ld = LaunchDescription(launch_args)
    ld.add_action(opfunc)
    return ld

**estimator_config**

%YAML:1.0 # need to specify the file type at the top!

verbosity: "DEBUG" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 2 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
[... 133 lines cut ...]
  intrinsics: [436.1545517976996, 436.20480087194466, 322.6650175012526, 233.28224614306288]
  resolution: [640, 480]
  rostopic: /a13/camera/right/image_raw
  timeshift_cam_imu: 1.4990728190294829
cam1:
  T_cam_imu:
    - [0.001469171404057723, 0.9996135264065523, 0.02776039193603369, 0.09049064226626044]
    - [-0.0016035446369905282, -0.02775803122841583, 0.9996133854380499, 0.0036551297174852015]
    - [0.9999976350871949, -0.001513118428608684, 0.0015621436036186986, 0.004876166919817045]
    - [0.0, 0.0, 0.0, 1.0]
  T_cn_cnm1:
    - [0.9999926029519827, -0.0036376372322886844, -0.001249654626396775, 0.09072058978617707]
    - [0.0036299819161040177, 0.9999749612432031, -0.00607454672745451, 0.002503486528430149]
    - [0.00127172033394357, 0.006069965570045485, 0.9999807689377688, 0.0038896121730224546]
    - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [0]
  camera_model: pinhole
  distortion_coeffs: [-0.3268054095693605, 0.09667096634166004, 0.0004165788885022756, -0.001039686437781763]
  distortion_model: radtan
  intrinsics: [437.7412328519853, 437.4565377431363, 318.0961775981548, 246.5198957501857]
  resolution: [640, 480]
  rostopic: /a13/camera/left/image_raw
  timeshift_cam_imu: 1.5000136263709618

[Screencast from 05-01-2024 08:16:42 PM.webm](https://github.com/rpng/open_vins/assets/135896050/31987ef7-988e-492e-b96b-3cea72807ad9)

goldbattle, 2024-05-03:
You are correct that theses config variables should be tuned. I would first get your console logging working.
Does adding this to your node show anything?
```
    output='screen',
    emulate_tty=True,
```

https://answers.ros.org/question/332829/no-stdout-logging-output-in-ros2-using-launch/

VivekMangeUD, 2024-05-03:
The snippet didn't work well but I used the snippet from the ROS answers and it worked to get the information on the terminal.

    **output='screen',
    emulate_tty=True,**

I have shared the window output and also the recordings of the test.
Recording link : https://drive.google.com/file/d/1m4vtrllglC6UgjXBH550nz2rtp1pWb2D/view?usp=sharing

**Update Launch file**

def launch_setup(context):
    configs_dir=os.path.join(get_package_share_directory('ov_msckf'),'config')
    available_configs = os.listdir(configs_dir)
    config = LaunchConfiguration('config').perform(context)
    if not config in available_configs:
        return[LogInfo(msg='ERROR: unknown config: \'{}\' - Available configs are: {} - not starting OpenVINS'.format(config,', '.join(available_configs)))]
    config_path = os.path.join(get_package_share_directory('ov_msckf'),'config',config,'estimator_config.yaml')
    node1 = Node(package = 'ov_msckf',
                 executable = 'run_subscribe_msckf',
                 namespace = LaunchConfiguration('namespace'),
                 **output={
                'stdout': 'screen',
                'stderr': 'screen'},**
                 parameters =[{'verbosity': LaunchConfiguration('verbosity')},
                              {'use_stereo': LaunchConfiguration('use_stereo')},
                              {'max_cameras': LaunchConfiguration('max_cameras')},
                              {'config_path': config_path}])
    return [node1]

[ros_openvins_terminal_output_log.txt](https://github.com/rpng/open_vins/files/15203236/ros_openvins_terminal_output_log.txt)

**Also, Can you help me with some information about the initialization process to better understand how it works and what different motion should I perform to trigger it or what params should I tune to get it triggered.**

VivekMangeUD, 2024-09-30:
The system is working now.

The issue was the time difference between IMU and Camera message. The system had a time dfifference of 1sec

closed 2024-09-30 by VivekMangeUD